chipsalliance/firrtl #299: Mem annotation crash

aswaterman, 2016-09-14:
Looks like it can't handle nested whens.  The following test case produces a stack overflow:

  circuit HastiTestSRAM :
    module HastiTestSRAM :
      input clk : Clock
      input hsel : UInt<1>
 
      reg p_valid : UInt<1>, clk
      reg p_address : UInt<5>, clk
      smem mem : UInt<8>[8][32] @[hasti.scala 517:19]
      when hsel : @[hasti.scala 524:16]
        when p_valid : @[hasti.scala 525:20]
          write mport T_155 = mem[p_address], clk

    Exception in thread "main" java.lang.StackOverflowError
      at sun.nio.cs.UTF_8$Encoder.encodeLoop(UTF_8.java:619)
      at java.nio.charset.CharsetEncoder.encode(CharsetEncoder.java:561)
      at sun.nio.cs.StreamEncoder.implWrite(StreamEncoder.java:271)
      at sun.nio.cs.StreamEncoder.write(StreamEncoder.java:125)
      at java.io.OutputStreamWriter.write(OutputStreamWriter.java:207)
      at java.io.BufferedWriter.flushBuffer(BufferedWriter.java:129)
      at java.io.PrintStream.write(PrintStream.java:526)
      at java.io.PrintStream.print(PrintStream.java:669)
      at java.io.PrintStream.println(PrintStream.java:823)
      at scala.Console$.println(Console.scala:148)
      at scala.Predef$.println(Predef.scala:310)
      at firrtl.passes.AnalysisUtils$.getOrigin(AnnotateMemMacros.scala:74)
      at firrtl.passes.AnalysisUtils$.getConnectOrigin(AnnotateMemMacros.scala:48)
      at firrtl.passes.AnalysisUtils$.getOrigin(AnnotateMemMacros.scala:75)
      at firrtl.passes.AnalysisUtils$.getConnectOrigin(AnnotateMemMacros.scala:48)
      at firrtl.passes.AnalysisUtils$.getOrigin(AnnotateMemMacros.scala:75)
      at firrtl.passes.AnalysisUtils$.getConnectOrigin(AnnotateMemMacros.scala:48)

shunshou, 2016-09-14:
I'm looking into it

shunshou, 2016-09-14:
It's not nested whens that it got stuck on. It was getting stuck on something stupider. ~_~ Basically, I check that whatever is connected to the memory enable signal equals the write mask. If enable signal is connected to Mux(p_valid, 1, 0) for example, then enable = p_valid. However, if p_valid should really be something else (due to constant propagation that hasn't occurred), I need to check for that.

The thing I didn't consider is what happens if p_valid <= p_valid [due to the register]... then my code goes in circles :( so I've cased that out.

aswaterman, 2016-09-14:
ha, makes sense

donggyukim, 2016-09-14:
Can be closed with #300?

shunshou, 2016-09-14:
Not yet. I'm still running into problems with Rocket [old version without Andrew's updates], even though I got that code snippet to work. Let me check to see what I'm doing wrong. Sorry!

shunshou, 2016-09-14:
Ok so Rocket presents an extended version of Andrew's snippet:

p_valid <= GEN_76
GEN_76 <= mux(T_159, UInt<1>("h1"), GEN_75)
GEN_75 <= mux(T_144, UInt<1>("h0"), p_valid)

Where I'm just trying to find the origin of p_valid, but it loops back to itself, not through a Connect, but through a Mux.

shunshou, 2016-09-14:
Ok the solution is to make a list of registers first and stop searching if a node = a register.

shunshou, 2016-09-15:
Ok, #301 should fix the version of rocket without Andrew's work around... Sorry, got distracted.

donggyukim, 2016-09-19:
By #301.

closed 2016-09-19 by donggyukim