chipsalliance/firrtl #308: Spec wrong for resultant bits of add

chick, 2016-09-20:
Firrtl spec says in section 7.1
```
add(UInt(w1), SInt(w2)) should result in SInt(w1.max(w2)+1)
```
But consider
```3.U<2> + 1.S<2>``` should have the numeric result +4 requiring 3 bits but because the result needs a zero sign bit so 4 can be positive, 
the actual SInt requires 4 bits or ```SInt(w1.max(w2)+1)```.
This is causing the interpreter to blow up.  Should the spec be changed, or if not what should the interpreter do here.

aswaterman, 2016-09-20:
Largest positive value is 2^w1 + 2^(w2-1) - 2, so max(w1, w2-1)+1+1 should suffice.

Largest negative value is -2^(w2-1), which is already dominated by the previous expression.

chick, 2016-09-20:
Where w2 is always the width of the SInt for both cases ```SInt + UInt``` and ```UInt + SInt```

aswaterman, 2016-09-20:
Yeah, that follows from commutativity

chick, 2016-09-22:
@jackkoenig @azidar, any plan on this?  This issue and #193 are causing users problems in the field when running the interpreter.  I don't think I can fix this in the interpreter without it agreeing with firrlt's behavior in this situation.  I think the fixes are fairly easy, plus an update to the manual

commit 24ffde94 (2016-09-22): Fixed width inference for add, sub (#312)

Fixes #308 
Fixes #193
closed 2016-09-22 by jackkoenig