chipsalliance/firrtl #311: DC Warning about Signed to Unsigned Assignment

palmer-dabbelt, 2016-09-22:
````
gn 'RecFNToRecFN_3'. (HDL-193)
Warning:  /scratch/palmer.dabbelt/work/plsi-nda-aspire-saed32/plsi/obj/soc-DefaultConfig-default/TestHarness.v:41451: signed to unsigned assignment occurs. (VER-318)
Warning:  /scratch/palmer.dabbelt/work/plsi-nda-aspire-saed32/plsi/obj/soc-DefaultConfig-default/TestHarness.v:41464: signed to unsigned assignment occurs. (VER-318)
Presto compilation completed successfully.
Information: Building the design 'DivSqrtRecF64_mulAddZ31'. (HDL-193)
Warning:  /scratch/palmer.dabbelt/work/plsi-nda-aspire-saed32/plsi/obj/soc-DefaultConfig-default/TestHarness.v:44166: signed to unsigned assignment occurs. (VER-318)
````

Here's a snippit (with line numbers) of the generated Verilog

````
 41432   assign T_8 = T_4[11:10];
 41433   assign T_10 = T_8 == 2'h3;
 41434   assign T_18_sign = T_25;
 41435   assign T_18_isNaN = T_27;
 41436   assign T_18_isInf = T_31;
 41437   assign T_18_isZero = T_7;
 41438   assign T_18_sExp = T_32;
 41439   assign T_18_sig = T_40;
 41440   assign T_25 = io_in[64];
 41441   assign T_26 = T_4[9];
 41442   assign T_27 = T_10 & T_26;
 41443   assign T_30 = T_26 == 1'h0;
 41444   assign T_31 = T_10 & T_30;
 41445   assign T_32 = {1'b0,$signed(T_4)};
 41446   assign T_35 = T_7 == 1'h0;
 41447   assign T_36 = io_in[51:0];
 41448   assign T_38 = {T_36,2'h0};
 41449   assign T_39 = {1'h0,T_35};
 41450   assign T_40 = {T_39,T_38};
 41451   assign T_42 = $signed(T_18_sExp) + $signed(13'sh1900);
 41452   assign outRawFloat_sign = T_18_sign;
 41453   assign outRawFloat_isNaN = T_18_isNaN;
 41454   assign outRawFloat_isInf = T_18_isInf;
 41455   assign outRawFloat_isZero = T_18_isZero;
 41456   assign outRawFloat_sExp = T_71;
 41457   assign outRawFloat_sig = T_76;
 41458   assign T_57 = $signed(T_42) < $signed(14'sh0);
 41459   assign T_58 = T_42[12:9];
 41460   assign T_60 = T_58 != 4'h0;
 41461   assign T_68 = T_42[8:0];
 41462   assign T_69 = T_60 ? 9'h1fc : T_68;
 41463   assign T_70 = {T_57,T_69};
 41464   assign T_71 = $signed(T_70);
 41465   assign T_72 = T_18_sig[55:30];
 41466   assign T_73 = T_18_sig[29:0];
 41467   assign T_75 = T_73 != 30'h0;
 41468   assign T_76 = {T_72,T_75};
 41469   assign T_77 = outRawFloat_sig[24];
 41470   assign T_79 = T_77 == 1'h0;
 41471   assign invalidExc = outRawFloat_isNaN & T_79;
 41472   assign RoundRawFNToRecFN_1_clk = clk;
 41473   assign RoundRawFNToRecFN_1_reset = reset;
 41474   assign RoundRawFNToRecFN_1_io_invalidExc = invalidExc;
 41475   assign RoundRawFNToRecFN_1_io_infiniteExc = 1'h0;
 41476   assign RoundRawFNToRecFN_1_io_in_sign = outRawFloat_sign;
 41477   assign RoundRawFNToRecFN_1_io_in_isNaN = outRawFloat_isNaN;
 41478   assign RoundRawFNToRecFN_1_io_in_isInf = outRawFloat_isInf;
 41479   assign RoundRawFNToRecFN_1_io_in_isZero = outRawFloat_isZero;
 41480   assign RoundRawFNToRecFN_1_io_in_sExp = outRawFloat_sExp;
 41481   assign RoundRawFNToRecFN_1_io_in_sig = outRawFloat_sig;
 41482   assign RoundRawFNToRecFN_1_io_roundingMode = io_roundingMode;
````

and here's a relevant-looking declaration of

````
 41378   wire [13:0] T_42;
````

@albert-magyar I haven'd read the Verilog LRM, but I'm assuming something like

````
-41451   assign T_42 = $signed(T_18_sExp) + $signed(13'sh1900);
+41451   assign T_42 = $(unsigned ($signed(T_18_sExp) + $signed(13'sh1900)));
````

would fix the problem?  Note that this is DC, so it could just be flat-out wrong.

aswaterman, 2016-09-22:
This isn't a bug, but I agree we should strive to generate lint-free code.
Curiously, VCS is fine with this; I guess Presto is more persnickety.  I
believe wrapping the result of signed expressions with $unsigned() is a
correct workaround.

ben-k, 2016-09-22:
Agree with @aswaterman about the fix.  And yes, Presto is stricter than VCS in syntax checking.

azidar, 2017-03-16:
I'm assuming this issue has been resolved - please reopen if it hasn't.

closed 2017-03-16 by azidar