chipsalliance/firrtl #320: Add max legal width

jackkoenig, 2016-09-26:
Inspired by #297 

We should enforce a max legal width of wires. It appears that Verilator doesn't support widths over 1 billion:
```
Width of bit range is huge; vector of over 1billion bits: 0x40000000
```
This might be a reasonable place to start

aswaterman, 2016-09-26:
Agreed.  As with Verilog, I don't think there's a compelling reason to make it part of the language spec; can just make it an implementation constant.

sdtwigg, 2016-09-26:
As long as the error messaging for dynamic left shift plays nice with this, it all seems fine. (Especially since << on a 32-bit register will trigger it almost every time). It seems like that is the most common pattern where this will come up anyway.

Basically, if the large wire is because of an inferred width being > 1 billion, the error message should be "Result of a << operation was >1 billion bits. Error." not "wire T65 is >1 billion bits. Error." The latter is, of course, acceptable in other cases such as the user trying to actually instantiate such a large signal.

jackkoenig, 2016-09-26:
Agreed, in general, since the dshl check in #318 and another check for max width in general will both be in the CheckWidths pass, they will both trigger at the same time so the additional information will be there.

jackkoenig, 2016-11-08:
Add to CheckWidths, pick same width as Verilator (and comment as much)

PR #403 (2016-12-13): Max width
Add MaxWidth of 1,000,000 bits

Also Move CheckWidths to its own file
Also base max dshl check on MaxWidth instead of just 31 bits
Resolves #320

This should be kept as 2 separate commits to actually be able to look at CheckWidths diff
commit 1ea7b6d6 (2016-12-13): Add MaxWidth of 1,000,000 bits

Also base max dshl check on MaxWidth instead of just 31 bits
Resolves #320
closed 2016-12-13 by jackkoenig