chipsalliance/firrtl #335: revisit width inference on ports

aswaterman, 2016-10-11:
This isn't a new issue, but it has come up again in conversation, so I'm revisiting it.  It continues to strike me as problematic that width inference problems on module ports are treated globally, rather than each instance being treated independently.

The main issue is composability.  Suppose module A instantiates module B, which has unspecified input widths, and the system functions properly.  Then, a new, unrelated module C instantiates B, but assigns signals of different width to B's input ports.  Module A's behavior consequently changes.  So even though A and C are unrelated, they cannot compose.

Notably, if module C were in a different circuit, which was later ganged together with A outside of FIRRTL, the circuits would compose.

It's not clear to me, OTOH, what advantage there is from instances of B affecting each other, aside from convenience of implementation.

sdtwigg, 2016-10-11:
I find the current behavior absurd. It is possible to mostly avert the issue by disabling deduplication in chisel. However, you are still susceptible to pre-width inference FIRRTL passes (or, I suppose any FIRRTL pass that invoked width inference) improperly instantiating a second instance of a module def instead of copying the module def.

Also, with that fix, it somewhat unnecessarily bloats FIRRTL files (and then, FIRRTL doesn't have any de-duplication logic in its emitter). I suppose the Chisel emitter could only duplicate module defs that have uncertainty in the input. Although, this require a slight complication of the Emitter and you won't get deduplication in the case where the instances of those defs all end up used the same.

azidar, 2016-10-11:
> The main issue is composability. Suppose module A instantiates module B, which has unspecified input widths, and the system functions properly. Then, a new, unrelated module C instantiates B, but [...]

I agree, this is certainly an issue. I think the problem stems from how Firrtl views a module. In Firrtl's worldview, a module is something that is not parameterized - all instances of a module are in fact identical. This makes, as you say, the implementation (and defining the behavior of width inference) simpler.

That being said, it does create this unintuitive behavior because, from Chisel's point of view, modules are parameterized by their widths. This causes the aforementioned composability issue, which is very undesirable. I'm happy to revisit this and consider changing the behavior.

To me, the biggest question is how the module uniquification becomes standardized in the Firrtl spec. If uninferred widths are allowed in ports, we would need to uniquify modules whose instances infer differing widths. What should the module renaming scheme be, such that is deterministic? 

Another complication is when some instances are unique, and some aren't? If I have the case where instance a and b of module M infer the same width, but instance c of module M infers a different width, what is the naming scheme under that?

aswaterman, 2016-10-12:
I think there are many justifiable answers, but I agree it's an important question to resolve, and the answer needs to be part of the spec.

It might be possible to generate a name that is then disambiguated using the existing signal-name disambiguator, which will simplify the description of one part of the solution.

azidar, 2016-10-12:
Here's another thought. To do proper width inference, either Firrtl or Chisel is going to need to uniquify modules that have ports of unknown width. Since Chisel _already_ has all modules uniquified, why doesn't Chisel just not deduplicate modules that have unknown widths? Firrtl can later do a deduplication pass (which is useful regardless of this use case). This allows Firrtl to have the simpler implementation and specification, while preserving Chisel's use case.

aswaterman, 2016-10-12:
I have contemplated that option but consider it a workaround, because it seems more useful to fix it in Firrtl (e.g. to cope with modules coming from other sources).  Regardless, it might be the best thing to do in the mean time, until this issue is firmly resolved.