chipsalliance/firrtl #345: FIRRTL generating broken bit-masked sequential memories

ccelio, 2016-10-21:
Here's my Chisel code:

```scala
val h_table = SeqMem(num_entries, Vec(fetch_width, Bool()))                    
...                                          
                                                                           
   val waddr = hwq.io.deq.bits.index                                           
   val wmask = hwq.io.deq.bits.executed                                        
   val wdata = hwq.io.deq.bits.takens                                          
   h_table.write(waddr, wdata, wmask)                                          
...                                                                      
```

Which turns into perfectly acceptable FIRRTL:
````verilog
smem h_table : UInt<1>[2][32768] @[2bc-table.scala 164:24]                                   
...                                             
  write mport T_99 = h_table[hwq.io.deq.bits.index], hwq.clock                               
  when hwq.io.deq.bits.executed[0] :                                                         
    T_99[0] <= hwq.io.deq.bits.takens[0]                                                     
    skip                                                                                     
  when hwq.io.deq.bits.executed[1] :                                                         
    T_99[1] <= hwq.io.deq.bits.takens[1]                                                     
    skip                                                                                     
  skip @[2bc-table.scala 172:4]                                                              

````

But then gets turned into the broken Verilog code in which it swaps the bit-mask bits and the write-data bits:

````verilog
module h_table(                                            
  input  [14:0] RW0_addr,                                  
  input   RW0_en,                                          
  input   RW0_clk,                                         
  input   RW0_wmode,                                       
  input   RW0_wdata_0,                                     
  input   RW0_wdata_1,                                     
  output  RW0_rdata_0,                                     
  output  RW0_rdata_1,                                     
  input   RW0_wmask_0,                                     
  input   RW0_wmask_1                                      
);                                                         
  wire [14:0] h_table_ext_RW0_addr;                        
  wire  h_table_ext_RW0_en;                                
  wire  h_table_ext_RW0_clk;                               
  wire  h_table_ext_RW0_wmode;                             
  wire [1:0] h_table_ext_RW0_wdata;                        
  wire [1:0] h_table_ext_RW0_rdata;                        
  wire [1:0] h_table_ext_RW0_wmask;                        
  h_table_ext h_table_ext (                                
    .RW0_addr(h_table_ext_RW0_addr),                       
    .RW0_en(h_table_ext_RW0_en),                           
    .RW0_clk(h_table_ext_RW0_clk),                         
    .RW0_wmode(h_table_ext_RW0_wmode),                     
    .RW0_wdata(h_table_ext_RW0_wdata),                     
    .RW0_rdata(h_table_ext_RW0_rdata),                     
    .RW0_wmask(h_table_ext_RW0_wmask)                      
  );                                                       
  assign RW0_rdata_0 = h_table_ext_RW0_rdata[1];           
  assign RW0_rdata_1 = h_table_ext_RW0_rdata[0];           
  assign h_table_ext_RW0_addr = RW0_addr;                  
  assign h_table_ext_RW0_en = RW0_en;                      
  assign h_table_ext_RW0_clk = RW0_clk;                    
  assign h_table_ext_RW0_wmode = RW0_wmode;                
  assign h_table_ext_RW0_wdata = {RW0_wdata_0,RW0_wdata_1}; /// <<---- !!!
  assign h_table_ext_RW0_wmask = {RW0_wmask_1,RW0_wmask_0}; /// <<---- !!!
endmodule                                                  
````

This kills the processor.

shunshou, 2016-10-21:
Whoa I assume this wasn't there before? I can fix it tonight w/ some other refactoring I intend to do.

ccelio, 2016-10-21:
Thanks, I really appreciate it.

> Whoa I assume this wasn't there before? 

I don't really pay attention to when my FIRRTL version gets bumped, so I don't know how long this has been around (BOOM gets bumped whenever Rocket-chip gets bumped).  It shows up in the branch predictor, so I only noticed this by analyzing the performance and seeing a regression.

shunshou, 2016-10-22:
lol weird broken edge case:

```
circuit Top:
  module Top:
    input clk: Clock
    input ptr : UInt<5>
    input mask: UInt<1>[2]
    input data: UInt<1>[2]

    smem h_table : UInt<1>[2][32]                                                                          
    write mport T_99 = h_table[ptr], clk                               
    when mask[0] :                                                         
      T_99[0] <= data[0]                                                     
      skip                                                                                     
    when mask[1] :                                                         
      T_99[1] <= data[1]                                                     
      skip                                                                                     
    skip  
```

Generates

```
  assign h_table_ext_W0_data = {W0_data_0,W0_data_1};
  assign h_table_ext_W0_mask = {W0_mask_1,W0_mask_0};
```

---

Whereas

```
circuit Top:
  module Top:
    input clk: Clock
    input ptr : UInt<5>
    input mask: UInt<1>[2]
    input data: UInt<5>[2]

    smem h_table : UInt<5>[2][32]                                                                          
    write mport T_99 = h_table[ptr], clk                               
    when mask[0] :                                                         
      T_99[0] <= data[0]                                                     
      skip                                                                                     
    when mask[1] :                                                         
      T_99[1] <= data[1]                                                     
      skip                                                                                     
    skip      
```

Generates the correct:

```
  assign h_table_ext_W0_data = {W0_data_0,W0_data_1};
  assign h_table_ext_W0_mask = {W0_mask_0,W0_mask_1};
```

Should be a bug in MemUtils. Will fix.

ccelio, 2016-10-22:
Thanks for working on this. 

Unfortunately, I'm having trouble testing this.   I've checked out the fix_bitmask branch on my firrtl submodule in rocketchip, and executed "sbt assembly" within the top directory, but I'm getting errors on trying to build rocket-chip with it:

```
Error: Unknown option --inferRW
Error: Unknown argument 'TestHarness'
Error: Unknown option --replSeqMem
Error: Unknown option --inferRW
Error: Unknown argument 'TestHarness'
Error: Unknown option --replSeqMem
Error: Unknown option -c:TestHarness:-o:/scratch/celio/boom-chip/vsim/generated-src/rocketchip.BOOMConfig.conf
Try --help for more information.
```

ccelio, 2016-10-22:
Manually applying your fix to the `hiermask` function to my copy of FIRRTL does indeed generate the correct code.  ^^

````verilog
assign h_table_ext_RW0_wdata = {RW0_wdata_0,RW0_wdata_1}; 
assign h_table_ext_RW0_wmask = {RW0_wmask_0,RW0_wmask_1}; 
````

shunshou, 2016-10-22:
Haha ok... yeah, I think rocket-chip firrtl is too far behind master (what with all of the new execute harness changes...)

colinschmidt, 2016-10-22:
you should look at rocket-chip/#409 for how to fix it.

donggyukim, 2016-10-24:
by #346, #349

closed 2016-10-24 by donggyukim