chipsalliance/firrtl #350: FIRRTL Match Error when building in vsim

zhemao, 2016-10-24:
I tried compiling the unittests in vsim on the latest rocket-chip master and got an internal FIRRTL exception.

    java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp /scratch/zhemao/rocket-chip/firrtl/utils/bin/firrtl.jar firrtl.Driver -i /scratch/zhemao/rocket-chip/vsim/generated-src/unittest.UncoreUnitTestConfig.fir -o /scratch/zhemao/rocket-chip/vsim/generated-src/unittest.UncoreUnitTestConfig.v -X verilog --infer-rw TestHarness --repl-seq-mem -c:TestHarness:-o:/scratch/zhemao/rocket-chip/vsim/generated-src/unittest.UncoreUnitTestConfig.conf
    Exception in thread "main" scala.MatchError: UnknownType (of class firrtl.ir.UnknownType$)
        at firrtl.Utils$.create_exps(Utils.scala:104)
        at firrtl.Utils$.create_exps(Utils.scala:95)
        at firrtl.passes.LowerTypes$.lowerTypesStmt(LowerTypes.scala:187)
        at firrtl.passes.LowerTypes$$anonfun$1.apply(LowerTypes.scala:145)
        at firrtl.passes.LowerTypes$$anonfun$1.apply(LowerTypes.scala:145)
        at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
        at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
        at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
        at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
        at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
        at scala.collection.AbstractTraversable.map(Traversable.scala:104)
        at firrtl.ir.Block.mapStmt(IR.scala:237)
        at firrtl.Mappers$StmtMagnet$$anon$1.map(Mappers.scala:41)
        at firrtl.Mappers$StmtMap$.map$extension(Mappers.scala:55)
        at firrtl.passes.LowerTypes$.lowerTypesStmt(LowerTypes.scala:145)
        at firrtl.passes.LowerTypes$$anonfun$1.apply(LowerTypes.scala:145)
        at firrtl.passes.LowerTypes$$anonfun$1.apply(LowerTypes.scala:145)
        at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
        at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
        at scala.collection.immutable.List.foreach(List.scala:381)
        at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
        at scala.collection.immutable.List.map(List.scala:285)
        at firrtl.ir.Block.mapStmt(IR.scala:237)
        at firrtl.Mappers$StmtMagnet$$anon$1.map(Mappers.scala:41)
        at firrtl.Mappers$StmtMap$.map$extension(Mappers.scala:55)
        at firrtl.passes.LowerTypes$.lowerTypesStmt(LowerTypes.scala:145)
        at firrtl.passes.LowerTypes$$anonfun$lowerTypes$1.apply(LowerTypes.scala:217)
        at firrtl.passes.LowerTypes$$anonfun$lowerTypes$1.apply(LowerTypes.scala:217)
        at firrtl.ir.Module.mapStmt(IR.scala:460)
        at firrtl.Mappers$ModuleMagnet$$anon$11.map(Mappers.scala:115)
        at firrtl.Mappers$ModuleMap$.map$extension(Mappers.scala:125)
        at firrtl.passes.LowerTypes$.lowerTypes(LowerTypes.scala:217)
        at firrtl.passes.LowerTypes$$anonfun$8.apply(LowerTypes.scala:221)
        at firrtl.passes.LowerTypes$$anonfun$8.apply(LowerTypes.scala:221)
        at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
        at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
        at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
        at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
        at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
        at scala.collection.AbstractTraversable.map(Traversable.scala:104)
        at firrtl.passes.LowerTypes$.run(LowerTypes.scala:221)
        at firrtl.SimpleRun$$anonfun$1$$anonfun$2.apply(LoweringCompilers.scala:47)
        at firrtl.SimpleRun$$anonfun$1$$anonfun$2.apply(LoweringCompilers.scala:47)
        at firrtl.Utils$.time(Utils.scala:56)
        at firrtl.SimpleRun$$anonfun$1.apply(LoweringCompilers.scala:47)
        at firrtl.SimpleRun$$anonfun$1.apply(LoweringCompilers.scala:45)
        at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
        at scala.collection.immutable.List.foldLeft(List.scala:84)
        at firrtl.SimpleRun$class.run(LoweringCompilers.scala:45)
        at firrtl.MiddleFirrtlToLowFirrtl.run(LoweringCompilers.scala:123)
        at firrtl.MiddleFirrtlToLowFirrtl.execute(LoweringCompilers.scala:132)
        at firrtl.Compiler$$anonfun$compile$1.apply(Compiler.scala:71)
        at firrtl.Compiler$$anonfun$compile$1.apply(Compiler.scala:70)
        at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
        at scala.collection.immutable.List.foldLeft(List.scala:84)
        at firrtl.Compiler$class.compile(Compiler.scala:70)
        at firrtl.VerilogCompiler.compile(LoweringCompilers.scala:202)
        at firrtl.Driver$.execute(Driver.scala:107)
        at firrtl.Driver$.execute(Driver.scala:130)
        at firrtl.Driver$.main(Driver.scala:145)
        at firrtl.Driver.main(Driver.scala)
make: *** [/scratch/zhemao/rocket-chip/vsim/generated-src/unittest.UncoreUnitTestConfig.conf] Error 1

Steps to reproduce in rocket-chip master.

    cd vsim
    make PROJECT=unittest CONFIG=UncoreUnitTestConfig

FIRRTL file is attached.
[unittest.UncoreUnitTestConfig.fir.txt](https://github.com/ucb-bar/firrtl/files/549222/unittest.UncoreUnitTestConfig.fir.txt)

jackkoenig, 2016-10-24:
Howie told me this works in emulator but not vsim which leads me to believe it's a bug in the memory blackboxing.

donggyukim, 2016-11-07:
by #352

closed 2016-11-07 by donggyukim