chipsalliance/firrtl #378: Nice names getting removed

jackkoenig, 2016-11-16:
Given simple Firrtl like the following:

```
circuit Mod :
  module Mod :
    input in : UInt<32>[4]                                                                                                                                                                                                                                                                                                  
    input idx : UInt<2>
    output out : UInt<32>
    node named = in[idx]
    out <= named
```

We lose the nice name "named" in favor of a generated name:

```verilog
module Mod(
  input  [31:0] in_0,
  input  [31:0] in_1,
  input  [31:0] in_2,
  input  [31:0] in_3,
  input  [1:0] idx,
  output [31:0] out
);
  wire [31:0] _GEN_0;
  wire [31:0] _GEN_1;
  wire [31:0] _GEN_2;
  wire [31:0] _GEN_3;
  assign out = _GEN_0;
  assign _GEN_0 = _GEN_3;
  assign _GEN_1 = 2'h1 == idx ? in_1 : in_0;
  assign _GEN_2 = 2'h2 == idx ? in_2 : _GEN_1;
  assign _GEN_3 = 2'h3 == idx ? in_3 : _GEN_2;
endmodule
```

It looks like it's being removed in DCE. I think this could easily be solved via a heuristic that names that start in `_` lose to names that don't. We could also introduce a "temp name" annotation that's inserted whenever a temp name is inserted and those always lose out to non-temp names.

jackkoenig, 2016-11-16:
Actually it's ConstantProp that is changing the `out <= named` connection to `out <= _GEN_0`, which then causes named to get DCEd. Same argument as above can be applied here.

jackkoenig, 2017-06-29:
With recent improvements to Constant Propagation, this problem is greatly exacerbated since intentionally named wires will now be removed as well as nodes. I believe that solving it in ConstProp should be relatively straightforward given that it already has backpropagation mechanisms. CSE is not so simple but could be refactored in a similar way.

Part of solving this problem is coming up with simple, predictable heuristics for how to choose names. The one I will start with is straightforward, always prefer something that doesn't begin with `_`, but I imagine there may be other desirable rules like prefer Wire to Node, etc.

mwachs5, 2017-06-29:
Just to add to the discussion, the problem is exacerbated even if the rule "keep named names over _T_ or _GEN names" is applied. 

Real world example is unpacking the fields from an AsyncResetReg (admittedly a messy hack we shouldn't have to keep around, but to be illustrative):

val FooWrData = Wire (new FooBundle())
val FooReg = Module(new AsyncResetReg(width = 32))
FooReg.io.d := FooWrData.asUInt()
val FooReadData = new FooBundle().fromUInt(FooReg.io.q)
...

In this example the name that is preserved even if you select "real" names over _T_ or _GEN_ may be  FooReg_io_q[31:0]. The bundle field unpacked names are lost. It's hard to make a rule about which are "more useful" or better to keep. It's sad to think there may be no way for a user to specify intermediate wires, as this seems to undo a lot of goodness that happened recently to make the Chisel-generated code human readable.

PR #620 (2017-06-30): Preserve "better" names in Constant Propagation
Names that do not start with '_' are "better" than those that do

Fixes #378 -- at least the original problem statement, as well as resolves how ConstProp was making good name removal much worse.

There is a reasonable debate to be had about how to pick between names that do not start with `_`. This does happen a bit in rocket-chip but most of them are replacing something like `in_0_d_bits_source` with `io_out_0_d_bits_source` although there are examples of things like `nackResponseMessage_size` being replaced by `probe_bits_size` or `pmp0_cfg_x` [... cut]
closed 2017-06-30 by jackkoenig