chipsalliance/firrtl #381: Module extractor

chick, 2016-11-21:
Can I get a script (implemented as a transform that extracts a given instance of a module from a firrtl file including any modules that it instantiates itself.
This would be very useful for rafael's validation work to be able to extract an internal module from a rocket chip firrtl file and run it by itself, using inputs driven by a vcd file

colinschmidt, 2016-11-21:
At the risk of adding work for myself...This already exists in https://github.com/ucb-bar/pfpmp GenerateTop is a transform that outputs verilog for only the module specified and modules it uses.

azidar, 2017-03-16:
Now this exists in https://github.com/ucb-bar/barstools. Please reopen if this does not meet your use case.

closed 2017-03-16 by azidar