chipsalliance/firrtl #389: FixedPoint and fromBits

grebe, 2016-11-30:
I'm not sure if this should be regarded as a chisel or firrtl problem. The following chisel
```
class A extends Module {
    val io = IO(new Bundle {
      val a = Output(Vec(1, UInt(32.W)))
      val b = Input( Vec(1, UInt(32.W)))
    })  

    val c = Wire(FixedPoint(32, 30))
    c.fromBits(io.b(0))
    io.a.fromBits(c)
  }
```

produces the following firrtl
```
circuit DumpVerilogA : 
  module DumpVerilogA : 
    input clock : Clock
    input reset : UInt<1>
    output io : {a : UInt<32>[1], flip b : UInt<32>[1]}
    
    io is invalid
    io is invalid
    wire c : Fixed<32><<30>> @[FFTSpec.scala 168:17]
    c is invalid @[FFTSpec.scala 168:17]
    wire _T_13 : Fixed<32><<30>>
    _T_13 is invalid
    _T_13 <= io.b[0] @[FFTSpec.scala 169:15]
    wire _T_16 : UInt<32>[1] @[FFTSpec.scala 170:18]
    _T_16 is invalid @[FFTSpec.scala 170:18]
    node _T_18 = bits(c, 31, 0) @[FFTSpec.scala 170:18]
    _T_16[0] <= _T_18 @[FFTSpec.scala 170:18]
```

I get a firrtl.passes.CheckTypes$InvalidConnect error on _T_13 <= io.b[0]. Should this be legal firrtl? Should fromBits() do something special in chisel? Perhaps FixedPoint should have a flatten that gives an SInt.

azidar, 2016-12-01:
No, this should not be legal firrtl, but instead fromBits() should use firrtl's asFixedPoint operator.

grebe, 2016-12-01:
Thanks. I filed ucb-bar/chisel3#391

closed 2016-12-01 by grebe