chipsalliance/firrtl #391: Deduplication Performance Bug

jackkoenig, 2016-11-30:
From #375 

Deduplication has polynomial time complexity

This PR seems to have an exponential effect on runtime. With the following code connecting `n` identical queues together:

```scala
class QueueTop(n: Int) extends Module {
  val io = IO(new Bundle {
    val in = Flipped(Decoupled(UInt(64.W)))
    val out = Decoupled(UInt(64.W))
  })

  var ref = io.in
  for (i <- 0 until n) {
    ref = Queue(ref, 8)
  }
  io.out <> ref
}
```

I got the following performance results (for chisel3 + Firrtl):

On firrtl master (9a967a27aa8bb51f4b62969d2889f9a9caa48e31):
```
For param 1: 43.6 ms
For param 2: 52.8 ms
For param 4: 83.3 ms
For param 8: 122.2 ms
For param 16: 174.2 ms
For param 32: 244.8 ms
For param 64: 417.3 ms
For param 128: 814.4 ms
```
Note that the resulting Verilog is *not* deduped (presumably because of the source locators which this PR is fixing).

On this branch (ab8440c045ef619db45f64e49263cbb08eea14a0):
```
For param 1: 51.0 ms
For param 2: 48.6 ms
For param 4: 80.1 ms
For param 8: 121.9 ms
For param 16: 379.8 ms
For param 32: 2443.0 ms
For param 64: 18830.0 ms
For param 128: (it was taking too long so I killed it)
```

Note at the end it's growth of 6.6x despite 2x the increase in the amount of work to do.
I will also point out that it does at least actually dedup.

Fixing this will allow https://github.com/ucb-bar/chisel3/pull/383 to be merged

colinschmidt, 2016-12-18:
Ok it turns out I do have an issue with this performance regression. I had only checked on the hwacha branch but the hurricane branch somehow falls off a cliff and doesn't finish (even given a few hours). The firrtl is about 5x bigger than before chisel dedupe was removed.

I can put the firrtl somewhere if you want but it seems like you have a good simple example here.

colinschmidt, 2017-01-11:
@jackkoenig Is this something that is on your TODO list? Or should I consider a fix to this to be a long way off?

jackkoenig, 2017-01-11:
I can look into it soon, is it causing you lots of problems?

colinschmidt, 2017-01-11:
I wouldn't say lots of problems but its preventing me from using a newer version of chisel and that includes annotation support. This isn't critical but would be nice to have.

commit 38b4850b (2017-01-18): Speed up deduplication. Fixes #391

Removes combining of source locators since the benefit is dubious.
Adds MultiInfo for faster appending of source locators in future.
commit 43189d84 (2017-01-19): Add MultiInfo. Speedup Info concatenation. Fixes #391
commit fd2f3218 (2017-01-19): Add MultiInfo. Speedup Info concatenation. Fixes #391
commit 962194c2 (2017-01-20): Add MultiInfo. Speedup Info concatenation. Fixes #391
closed 2017-01-20 by jackkoenig