chipsalliance/firrtl #414: Constant Prop Wires as well as Nodes

jackkoenig, 2017-01-19:
See https://github.com/ucb-bar/chisel3/issues/443

I believe this would remove a lot of the pointless assignments. We also might consider adding an annotation for temporary nodes so that we can use that information in choosing which names to remove. This idea relates to https://github.com/ucb-bar/firrtl/issues/378

azidar, 2017-03-16:
related to #490

sols1, 2017-03-30:
Another thing that would be useful is to eliminate multiple inversions of the same signal:

[image]

In this example:

```
_T_1795 = ~uncachedInFlight_0
_T_1797 = ~_T_1795
_T_1799 = ~_T1797
```

sols1, 2017-03-31:
Another case of a lot of intermediate `_T` signals that should be possible to fold:

[image]

No wonder it is hard to close toggle coverage :)

aswaterman, 2017-03-31:
Oof.

I think constant propagation through wires will help somewhat, as well optimizing the Extract(Extract(...)) case.

jackkoenig, 2017-12-15:
Original issue fixed by https://github.com/freechipsproject/firrtl/pull/684, will file new issue for const prop improvement

closed 2017-12-15 by jackkoenig