chipsalliance/firrtl #420: Other reasons turning on ReplSeqMem is broken?

shunshou, 2017-01-26:
So I've been meaning to diagnose @bwrcbcr 's problem with Legalize, but on a simpler test circuit. The one I used was generated directly from Chisel [Trying to get OptionsManagers to work is another story...]:

```
;buildInfoPackage: chisel3, version: 3.1-SNAPSHOT, scalaVersion: 2.11.7, sbtVersion: 0.13.11, builtAtString: 2017-01-26 12:23:07.403, builtAtMillis: 1485433387403
circuit CustomMemory : 
  module CustomMemory : 
    input clock : Clock
    input reset : UInt<1>
    output io : {flip rClk : Clock, flip rAddr : UInt<3>, dO : SInt<16>, flip wClk : Clock, flip wAddr : UInt<3>, flip wEn : UInt<1>, flip dI : SInt<16>}
    
    io is invalid
    io is invalid
    smem mem : SInt<16>[7] @[SRAMSpec.scala 42:19]
    read mport _T_17 = mem[io.rAddr], clock
    io.dO <= _T_17 @[SRAMSpec.scala 45:21]
    when io.wEn : @[SRAMSpec.scala 56:31]
      write mport _T_18 = mem[io.wAddr], clock
      _T_18 <= io.dI
      skip @[SRAMSpec.scala 56:31]
```

Only when I enable ReplSeqMem does the Verilog generation fail with 

```
scala.MatchError: (SIntType(IntWidth(16)),UIntType(IntWidth(16))) (of class scala.Tuple2)
[info]   at firrtl.passes.InferWidths$.firrtl$passes$InferWidths$$get_constraints_t$1(InferWidths.scala:224)
[info]   at firrtl.passes.InferWidths$$anonfun$firrtl$passes$InferWidths$$get_constraints_s$1$1.apply(InferWidths.scala:265)
[info]   at firrtl.passes.InferWidths$$anonfun$firrtl$passes$InferWidths$$get_constraints_s$1$1.apply(InferWidths.scala:263)
[info]   at scala.collection.TraversableLike$$anonfun$flatMap$1.apply(TraversableLike.scala:252)
[info]   at scala.collection.TraversableLike$$anonfun$flatMap$1.apply(TraversableLike.scala:252)
[info]   at scala.collection.immutable.List.foreach(List.scala:381)
[info]   at scala.collection.TraversableLike$class.flatMap(TraversableLike.scala:252)
[info]   at scala.collection.immutable.List.flatMap(List.scala:344)
[info]   at firrtl.passes.InferWidths$.firrtl$passes$InferWidths$$get_constraints_s$1(InferWidths.scala:263)
[info]   at firrtl.passes.InferWidths$$anonfun$firrtl$passes$InferWidths$$get_constraints_s$1$6.apply(InferWidths.scala:292)
```

Aside from the extra io is invalid line and the underscore preceding letters weirdness, I don't think I see anything out of place for this CHIRRTL... (although I'm not fully cognizant right now...)...

Has anyone tried running the transform lately?

colinschmidt, 2017-01-26:
The underscores are just to force verilator to ignore those signals. I'd guess there is some bad interplay between the pass and SInts? I've been running it often with a very new firrtl/a51e72fefe1d33a1077e85cf04e1750dad6f8704 and a reasonably new chisel3/ad53161bbb9f67e16b88ca7a508a537f88d77e05 with no issues. But I dont use SInts in my memories

shunshou, 2017-01-26:
Found the bug. Working on a fix.

shunshou, 2017-01-26:
@azidar the memory pass does a lot of bit manipulation to chunk up memory contents correctly (i.e. for a Vec of SInts, all of the SInts get concatenated together as the data into the memory). In order for this to work with anything except UInts, you need to do an explicit cast back to SInt, Fixed, etc. for the read out data. Should something like this mess around with range inference if range inference occurs later? (Seems like InferWidths happens in Middle to Low Firrtl, which is after ReplSeqMem). Or is it ok, b/c we're assuming ranges are fixed/defined @ module boundaries? i.e. input range == output range. But that info needs to definitely make it past this pass (worth another thought!)

PR #422 (2017-01-28): Fix signed types
@bwrcbcr this should fix your memory problems -- at least with the *.fir file you pointed me to. @azidar @jackkoenig can one of you review my PR ASAP? It's gating CRAFT. Also, there are other type matches in Legalize (prob elsewhere) that are missing Fixed... The purposes of those functions weren't immediately obvious to me, so I didn't try adding stuff. It'd be good if Adam went through and looked for any matches w/ SInts that indicate FixedType matches should also be done. 

Fixes #420 #419
closed 2017-01-28 by shunshou