chipsalliance/firrtl #426: Possible fixed point changes to consider?

shunshou, 2017-01-28:
Besides getting ranges to work so you can optimize multiply bits [i.e. don't go 2*w each time...], etc.

---

Say I have a fixed point # x = FixedPoint(8.W,4.BP)

If I do y = x >> 7, the result has W = 1, BP = 4, which apparently is legal (?) since, when I assign y to an output that's 8.W, 4.BP, basically, the sign bit is repeated 8 times. Basically, if you shift right by width - 1, for a negative number, you should get the smallest possible non-zero number representable. 

i.e. if you have 1.1 (W = 2, Integer Bits = 0, Fractional Bits = 1), shifting right by 1 will again result in 1.1, since the next half is unrepresentable between 1.1 and 0. 

However! What should be the behavior when you do x >> 8 (i.e. shift by the full width)? 

Currently, I get some FIRRTL exception:

```
firrtl.FIRRTLException: Unsupported type FixedType(IntWidth(4),IntWidth(4)) for Primop Shift Right
[info]   at firrtl.Utils$.error(Utils.scala:256)
[info]   at firrtl.passes.Legalize$.legalizeShiftRight(Passes.scala:179)
[info]   at firrtl.passes.Legalize$.firrtl$passes$Legalize$$legalizeE$1(Passes.scala:221)
```

Really, I think you should check to see if the shift right amount is >= width, and if so, set to 0. I'm not sure if this should be done in FIRRTL or done in Chisel...

---

Also,

```
case Cat => (t1, t2) match {
        case (_: UIntType | _: SIntType, _: UIntType | _: SIntType) => UIntType(PLUS(w1, w2))
        case (_: FixedType, _: UIntType| _: SIntType) => FixedType(PLUS(w1, w2), PLUS(p1, w2))
        case (_: UIntType | _: SIntType, _: FixedType) => FixedType(PLUS(w1, w2), p1)
        case (t1, t2) => UnknownType
      }
```

I disagree with this, because it's extremely confusing. I consider Cat a "bitwise" operation -- therefore the output type should always be UInt and require an explicit cast back to Fixed if needed. At least that's how I feel...

That's assuming that Cat isn't used to do anything like aligning binary points (but it seems like you use shr/shl to do that, which is implied to be arithmetic?)...

@azidar @grebe @chick opinions?

azidar, 2017-01-30:
Thanks for the writeup Angie. I remember when I was writing these, I wasn't sure the exact behavior that would be desired, so I'm glad you've thought some more about this.

RE: Cat. Sounds good, we can always return a UInt.
RE: Shr. Are you ok with shr not affecting the binary point position? And if I am understanding you correctly, for shifting a value by larger than its width, you want to return zero?

shunshou, 2017-01-30:
I *think* (haven't thought too hard...) that shr shouldn't affect the binary point. Basically, I want shr to function as a divide by 2^n, so if you move bits around, it should be relative to the binary point. And yes, I want to return zero. Probably should have @grebe or someone else chime in, but to me, that's the most logical.

shunshou, 2017-02-01:
Hm... you know, never mind about shift. I guess it should just stick with convention which is keep shifting in a 1 @ MSB, so it's always round towards negative infinity for negative numbers...

shunshou, 2017-02-01:
Actually, that's weird. I tried shifting by the width again, and it didn't fail this time? @_@ Not sure what I did earlier...

shunshou, 2017-02-01:
Ok, so I guess >> should probably behave the way you have it -- since it's consistent w/ SInt and general interpretations of >>. I made a separate div2 operation in DspTools that does:

```
   require(a.widthKnown, "Fixed point width must be known for div2")
    require(a.binaryPoint.known, "Binary point must be known for div2")
    val newBP = a.binaryPoint.get + n
    val inLong = Wire(FixedPoint((a.getWidth + n).W, newBP.BP))
    inLong := a
    val outFull = Wire(FixedPoint(a.getWidth.W, newBP.BP))
    outFull := inLong >> n
    outFull
```

That way, I maximize significant bits... Not sure if this should be in DspTools or in Firrtl or elsewhere...

closed 2017-02-08 by azidar