chipsalliance/firrtl #429: setBinaryPoint is misleading (?) -- maybe should be set precision? Firrtl/Chisel question...

shunshou, 2017-02-01:
Also, for operations with inputs of known widths/binary points, should Chisel's getWidth and binaryPoint.get return what Firrtl ends up doing? Or are there supposed to be discrepancies?

Say I have

val in = Input(FixedPoint(6.W, 4.BP))

If I try in.setBinaryPoint(8) and print out in.binaryPoint.get and in.getWidth, I get:

Width = 6, BP = 8 -- In this case, according to Chisel, you've cut off the MSBs, which *imo* is not "set binary point."

If I try in.setBinaryPoint(0), at least the output should be fully represented, according to Chisel:

Width = 6, BP = 0

Except that technically, you only need 2 bits to represent floor(in). 

If I look at the Verilog generated by Firrtl, however, 

If x = in.setBinaryPoint(8), it allocates 10 bits for x, which is right! (and then x = in << 4).
If x = in.setBinaryPoint(0), it allocates 2 bits for x. Again, consistent with just taking the int part.

I kind of assumed that if the input had KnownWidth and KnownBinaryPoint, as long as we checked to see that stuff was "Known", we'd get correct (at least to first order) results, but this is inconsistent?

In general, I find setBinaryPoint to be slightly misleading. If Firrtl does always expand to be able to represent the #, you're not really doing anything in terms of significant digits, and really, you're just expanding, shrinking the precision. 

@grebe @azidar @chick opinions?

azidar, 2017-02-08:
I think the idea was, in the future, we can support negative binary positions (though we aren't now). I don't know if a "negative precision" makes sense, but maybe it does. IMO I don't care about its name, so if you and others feel strongly about it, I'm fine changing it.

shunshou, 2017-02-10:
BinaryPoint name aside (and I'm not really big on needing negative precision...), I would like my first sentence addressed. @azidar Firrtl generates the correct output, but not with the same bitwidths that Chisel expects, even though things have KnownBinaryPoint. This is dangerous b/c any user code that checks KnownBinaryPoint and tries to do something with it might end up doing the *wrong* thing. 

This issue exists for setBinaryPoint, as mentioned above. I'm not sure if it exists for other ops; I'm not even sure really how to check for this w/o very manual visual inspection...

closed 2017-02-25 by shunshou