chipsalliance/firrtl #431: External Library Transforms

shunshou, 2017-02-07:
I don't think this has ever been properly addressed from the Chisel down to Firrtl level. The handful of use cases I see of an external library transform require that Firrtl be separately run from Chisel -- therefore, people can create their own CustomCompiler and pass that in to firrtl.Driver.compile. 

However, if you use the iotesters.Driver.execute or similar (w/o having to run Firrtl w/ a separate JVM), you can't feed in a compiler: Compiler option (although you can choose "verilog", etc.). 

I think the pad pass I want to write is a good candidate for an external library transform, but I'd really like to do it w/o running Firrtl separately -- granted, you'd still need to probably have the pass be in base Firrtl to have the CRAFT flavor of rocket-chip as @grebe and @stevobailey have dealt with it.

@chick is this something you can add? Should be quick. Also, do @azidar and co. think this should be an external pass or one native to Firrtl? IMO there should be some Tapeout specific library (which might be pmfmp or w/e it's called), but maybe for the purposes of CRAFT, it'll be too annoying to separate stuff out...

colinschmidt, 2017-02-07:
I personally would be happy if you could run custom firrtl compilers ala pfpmp in the same JVM(Assuming this would be a performance improvement). For my use case I would still want the firrtl file written out so just a little additional requirement.

A little off-topic but related to your last paragraph: I think ucb-bar|bwrc should strive to keep the passes it develops for tapeouts in one place. For I think just leaving it as ucb-bar/pfpmp is fine. I would hope our tapeout specific chisel code will live in ucb-bar/testchipip.

shunshou, 2017-02-08:
I think you actually can do that. You just need to separately have a Chisel driver -> dump file -> Firrtl driver -> read file + annotations -> output. I did this back many many months ago, and at least this was all written up in the same Scala file, so I assume it was a single JVM (back before a more formalized executions options manager existed). I just really dislike the notion of having to separately run Chisel driver and then Firrtl driver (even if it does this internally). Chisel.Driver.execute should pass all the appropriate annotations, compiler info down from the top. 

Also, why do you want the firrtl file output? Theoretically, the intermediate representation should not be important at all. I can understand that, from the perspective of debugging still-lurking Firrtl bugs this will be useful, but in "a perfect world," you shouldn't have to worry about the Chirrtl dump.

My issue with ucb-bar/testchipip is that it's still Rocket-chip centric, and although Rocket-chip is something like 90% of the Chisel ecosystem right now, I think it's a terrible idea to have all tapeout specific Chisel code live in ucb-bar/testchipip. 

Re: pad pass, the easiest thing to do is probably stick it directly in Firrtl at this point... I think you can have "one place" that aggregates passes/transforms (via submodules or the likes), but I'd personally like to develop stuff independently of what's in pfpmp.

Also, it's kind of annoying having to pass in technology specific stuff to a public repo (accidental exposure would have serious consequences), so maybe it'd be easier if it was hosted only on bwrcrepo.

colinschmidt, 2017-02-08:
Ok. Currently I'm not that concerned with exactly how I run chisel+firrtl but I just wanted to lend support to your issue.

I use the firrtl output to inspect for debugging as well as for some non-firrtl scripted changes because its much easier to deal with than verilog. Eventually these should be firrtl passes so I agree we won't need that feature long term.

So by tapeout specific I mean things like the tether, SCR files, etc. Putting them in one place helps us get reuse out of our designs, which means when I fix and SCR file bug your tapeouts can also have that bug fixed. I would be happy to move it away from rocket-chip but for the time being it seems to be the vast majority of our chisel tapeouts. 

I guess I'm not clear on what the pad pass is and what technology specific stuff is involved. So I'll defer to your judgment.

azidar, 2017-02-08:
TBH, I'm fine with adding more firrtl transformations to this repository, as long as its put in a separate package and directory. At some point in the future we can split it out, but I'm happy to co-locate firrtl passes so we can learn from each other and distribute the knowledge a bit better.

shunshou, 2017-02-10:
Ok I'll do that.

colinschmidt, 2017-02-10:
Ok so @chick just showed me that any annotations attached to chisel code will have their associated transforms automagically run during the firrtl execution! I'm not sure if this is what you want but it satisfies my use case excellently.

shunshou, 2017-02-10:
I'm not sure I understand what you mean? You mean passing annotations via the execution options manager? 

That doesn't work for transforms that don't exist in the firrtl package.

colinschmidt, 2017-02-10:
No just annotations that you've put in your chisel code.

I suppose that might be a limitation maybe @chick can comment on how that would work.

shunshou, 2017-02-10:
I totally believe that that's possible -- is there good example code (simple!) on how to deal with Annotations from Chisel all the way to FIRRTL?

shunshou, 2017-02-10:
There is https://github.com/ucb-bar/chisel3/wiki/Annotations%20Extending%20Chisel%20and%20Firrtl but can I just get an example of how to annotate a node with "fox" and have FIRRTL read that "fox" is attached to the node...

colinschmidt, 2017-02-10:
This https://github.com/ucb-bar/chisel3/blob/master/src/test/scala/chiselTests/AnnotatingDiamondSpec.scala has an annotation, and transform in the same file. The transform is automatically called

chick, 2017-02-10:
@shunshou I can walk you through this.   Obviously more doc needed here.

shunshou, 2017-02-10:
I'd definitely like to use that when I start working on the pad/top level pass.

closed 2017-03-16 by azidar