chipsalliance/firrtl #440: VCS is not happy with $fatal

wsong83, 2017-02-13:
Is there a way to get around of this or I am doing something stupid?

Basically I generate some verilog with Chisel assertions. So firrtl emit $fatal.
However, VCS complains that $fatal is a SV feature but used in a non-SV context.
I managed to pass around VCS compilation by changing Emitter.scala:

    	def stop(ret: Int): Seq[Any] = Seq(if (ret == 0) "$finish;" else "assert (0) else $fatal(1);")

Obviously this is a terrible hack as the fake assert makes no sense except for complying with SV standard.

aswaterman, 2017-02-13:
This isn't really an answer to your question, but did you consider adding the -sverilog flag to VCS?

wsong83, 2017-02-13:
Yes I did.

aswaterman, 2017-02-14:
That's odd, because we use VCS and have no problem with $fatal.

wsong83, 2017-02-14:
I see. I will try to narrow it down then. Close it for now.

closed 2017-02-14 by wsong83

wsong83, 2017-02-16:
FYI, it compiles after updating to latest VCS.

wsong83, 2017-03-17:
I still think it might be better to change it. $fatal is a SV feature and using $fatal like $finish (without a sesert) is a non-standard way of using it.

Besides the old VCS, now I just find it will cause Vivado simulation (xelab) silently generating a segment error.