chipsalliance/firrtl #446: Enforce ordering on custom pass annotations

colinschmidt, 2017-02-17:
When directly calling a firrtl compiler with custom transforms the user has the ability to enforce an order between their passes. If annotations are used to add custom transforms there is no extant way to order them. This would be a very useful feature.

shunshou, 2017-02-17:
One thing I might add, and not sure if anyone's really tried -- for ResetInverter, i assume the invert(this) thing could also be run from some higher module that instantiates a Module with ResetInverter (or whatever it was called) trait. If that's the case, people might be able to order them to some degree... but when dealing with transforms done on the top level module, I'm not sure how one goes about doing that with Chisel Driver? I'm still kind of confused about how the (): => stuff works...

shunshou, 2017-02-21:
As another example: I *need* my AddPads Transform to come before BlackBoxHelper, but I don't know if there's any way to guarantee that, given that both transforms should occur on LowForm.

azidar, 2017-05-10:
Current thoughts: list of passes required before/after running - topo sort, then insert lowering passes to fix form mismatches

seldridge, 2019-07-19:
Current approach in #1123 is that you get two ways of defining dependencies:

1. The custom `Transform`s can define prerequisites, dependents, and invalidates
2. The order of the `RunFirrtlTransformAnnotation`s is used in seeding the topological sort. This has the effect of transforms, with the same dependencies, running in the same order they're specified. This should match the current behavior. 

Implementation-wise for (2) explicit dependencies are not added based on the order.

closed 2020-03-11 by seldridge