chipsalliance/firrtl #448: Slow "common subexpression elimination" pass with large ROM design.

stevobailey, 2017-02-19:
My design has lots of constants, and it doesn't scale well with design size. Generating the FIRRTL is quick, but FIRRTL is very slow during the "common subexpression elimination" pass relative to all the other passes. I assume this is because it's trying to optimize a large ROM. Maybe this behavior is expected, but it would be nice to have a faster version, whether through better handling of this case or an option to turn off this pass.

The FIRRTL files and "log-level info" outputs for 4 design sizes are in /nscratch/stevo/firrtl-test. 

This can also be replicated by compiling the FFT yourself. Clone the [FFT repo](https://github.com/ucb-art/fft) (follow the setup instructions in the README), and change the design size ([here](https://github.com/ucb-art/fft/blob/master/src/main/scala/fft/Config.scala#L95)) to 256, 512, 1024, or 2048. Compile with `make verilog CONFIG=CustomStandaloneFFTConfig`. 

[image]

[image]

[image]

shunshou, 2017-02-19:
That's why I keep emphasizing we need a dedicated ROM construct (which I think @jackkoenig was going to work on at some point?) -- I would like to know if there's an estimated completion date, because 3-4 (time-consuming) things down on my TODO list is to black box ROMs in Chisel via a custom ROM module.

aswaterman, 2017-02-19:
There are two issues here.  One is calling out ROMs to support backend-specific code generation.  We have been using black boxes for this, and it works fine for our purposes.

The second issue is optimizing code generation for behavioral ROMs.  This could be pattern-matched and optimized entirely in FIRRTL, without adding new FIRRTL constructs or modifying Chisel.  If it's done early enough in FIRRTL compilation, it should sidestep the performance issues.  This matches the tried and true approach that software compilers take.

This all buries the lede.  I'm surprised no one pointed out the real problem here--the FIRRTL runtime appears to be super-linear in the ROM size.  THAT should be the first problem to attack.

shunshou, 2017-02-19:
Super-linear runtime aside, I think the the first 2 issues are kind of the same thing. There are certain causes where you might want to pattern match for a Vec of all Lits and optimize in Firrtl. There are other cases when you legitimately want to blackbox and run a foundry-specific ROM compiler. You'd need (or at least want?) some kind of annotation or other to be able to easily switch between the two possibilities.

shunshou, 2017-02-19:
Ok, I hadn't really looked into foundry ROM docs -- I just figured that in certain cases, they'd be useful, b/c the ROM would get its own dedicated (uniform) area, and that potentially helps w/ things like routing congestion, which seems to be a big deal as your ROM sizes increase. I added registers at the output of my ROM anyways, b/c the combinational logic delay was too high otherwise. That being said, maybe the *right* approach really is just to black box stuff from the get-go and don't rely on Chisel/Firrtl to do anything if you know your ROM is sufficiently large?

aswaterman, 2017-02-19:
I think the main reason is that dense ROMs have dynamic circuitry, like SRAMs, and so need an edge or a strobe.  The reason we've used them is they can be reprogrammed by just changing a metal layer, not for their resource advantages.  (I suspect their dynamic power is actually worse than combinational logic, if there is locality of reference; not sure about random access.)

In any case, I agree that the black-box approach is a conservative approach both to have implementation flexibility and to short-circuit the performance issues of large ROMs.  I don't think we should close the door on possible Chisel/FIRRTL support.

jackkoenig, 2017-02-19:
I think @azidar had floated the idea of a dedicated ROM construct. Regardless Firrtl most certainly should not be superlinear in the simple behavioral case. I'm looking into performance stuff and will look into this soon.

shunshou, 2017-02-19:
Maybe it'd be a useful exercise at some point to do a power/area comparison between combinational logic vs. compiled ROMs over some range of applications. I'm not necessarily talking about resource advantages either -- the tools should probably be better able to deal with a ROM macro rather than having to optimally place all the combinational logic associated with the (large) ROM. @stevobailey can tell you his stories of his ring of death or w/e, but, ease of getting stuff through tools is actually really valuable... at least for one-off tapeouts... ~_~

Ok, well, if no one plans to have a ROM Firrtl construct in place within the next <1 month, then black box approach it is. At least for me. Good to know.

aswaterman, 2017-02-19:
Agreed @shunshou - by "resources" I meant routing tracks, too

jackkoenig, 2017-02-20:
Right so the problem is 2 fold:

1. DirectFFT creates a lot of duplicate nodes that get CSEd. I believe the main culprit is: 
https://github.com/ucb-art/fft/blob/master/src/main/scala/fft/FFT.scala#L60

| FFT size | # of CSE runs for DirectFFT |
| --- | --- |
| 256 | 226 |
| 512 | 450 |
| 1024 | 898 |

The number of runs seems to scale linearly with the FFT size, but because the module is now bigger, it takes longer to CSE, thus the superlinear slowdown.

2. CSE rehashes all of the nodes in a module each time it runs

I can fix (2) by making the pass a little more clever which will be a big constant factor improvement. I'm not sure if there is anything Firrtl can do about (1), that is kind of a fundamental property of the design. Perhaps some cleverness can help but designs with lots of duplication will require more CSE.

stevobailey, 2017-02-20:
Okay, thanks for looking into this! So yes, using a ROM or two would help eliminate these nodes. I can look into design modifications if we end up using these large FFTs.

aswaterman, 2017-02-20:
A slight modification to that code would probably improve performance noticeably (and as a side effect improves clarity):

    twiddle := Vec((0 until lanesIn-1).map { case x =>
      val index = indices_rom(start+UInt(x))
      Mux(index(log2Ceil(config.n/4)),
        DspComplex.divideByJ(twiddle_rom(index(log2Ceil(config.n/4)-1, 0))),
        twiddle_rom(index))
    })

aswaterman, 2017-02-20:
or something like that, I didn't try compiling it.

stevobailey, 2017-02-20:
Thanks, I get a little overzealous in prioritizing one-liners over efficient code, haha.

aswaterman, 2017-02-21:
Yeah, that means we definitely need to fix CSE.

commit f7bb5939 (2017-04-13): Speed up CSE by doing CSE on node expression before recording the node

This means CSE need only be run once to get same QOR and prevents
pathological cases. Fixes #448
PR #543 (2017-04-13): Speed up CSE by doing CSE on node expression before recording the node
This means CSE need only be run once to get same QOR and prevents
pathological cases. Fixes #448 

Despite the long rabbit hole of looking at reordering wire connections and nodes, a small change fixes the pathological case.
commit 1c42e87b (2017-04-13): Speed up CSE by doing CSE on node expression before recording the node (#543)

This means CSE need only be run once to get same QOR and prevents
pathological cases. Fixes #448
closed 2017-04-13 by jackkoenig