chipsalliance/firrtl #450: shift right does not complain about shift size greater than width

chick, 2017-02-20:
The firrtl spec 7.12
> The shift right operation truncates the least significant n bits from e. 
> n must be non-negative and         strictly less than the bit width of e.

The following code, run with n = 8, 9, 10
```scala
class BrokenShifter(n: Int) extends Module {
  val io = IO(new Bundle {
    val i = Input(FixedPoint(8.W, 4.BP))
    val o = Output(FixedPoint(8.W, 4.BP))
    val si = Input(SInt(8.W))
    val so = Output(SInt(8.W))
  })
  io.o := io.i >> n
  io.so := io.si >> n
}
```
generates the following firrtl, which is processed without complaint.
```
circuit BrokenShifter : 
  module BrokenShifter : 
    input clock : Clock
    input reset : UInt<1>
    output io : {flip i : Fixed<8><<4>>, o : Fixed<8><<4>>, flip si : SInt<8>, so : SInt<8>}
    
    io is invalid
    io is invalid
    node _T_10 = shr(io.i, 9) @[FixedPointSpec.scala 135:16]
    io.o <= _T_10 @[FixedPointSpec.scala 135:8]
    node _T_11 = shr(io.si, 9) @[FixedPointSpec.scala 136:18]
    io.so <= _T_11 @[FixedPointSpec.scala 136:9]
    
```

chick, 2017-02-20:
I think this should cause firrtl to error, chisel3 may not know the width at elaboration so it cannot catch all cases.  I don't know if it is worth having it do so when the width is known.

sdtwigg, 2017-02-20:
it has actually been occasionally awkward that you cannot shift equal to the width (and just get 0).

shunshou, 2017-02-20:
^ for signed, you'd never get zero. Tho originally I was thinking that if you did want to shift > width, you'd get 0 regardless.

jackkoenig, 2017-02-20:
The current behavior is that right shifting an UInt right by >= width results in 0, while right shifting a SInt by >= width gives you the sign bit.

As we discussed in https://github.com/ucb-bar/firrtl/issues/69 and merged in #90, making shift right by more than the width of the argument illegal is annoying in generators (requires lots of special casing) and is inconsistent with dynamic shift right where you obviously can't error.

chick, 2017-02-20:
ok, seems like I should just update the firrtl spec

jackkoenig, 2017-02-20:
Yeah, we (I) have done a really bad job keeping it up to date, I need to update some stuff as well

shunshou, 2017-02-20:
@chick then tests for UInt, SInt, FixedPt for dsp stuff should also check for > width.

azidar, 2017-12-15:
Resolution: need update spec.

commit 74a3b302 (2018-02-16): Update spec for rhs

Fixes #450
closed 2018-02-16 by azidar