chipsalliance/firrtl #452: MemLib and Wiring should use standard annotations YAML reader/writer

colinschmidt, 2017-02-21:
Right now they have custom YAML which doesn't seem like a good idea for maintainability.

azidar, 2017-12-15:
New annotation serialization is coming soon, making this issue obsolete.

edcote, 2017-12-27:
@azidar is there a preview of this new API for annotation serialization?

jackkoenig, 2017-12-27:
I plan to make a PR this week or next. The gist is that Annotations that are case classes and only use standard Scala types (Strings, Ints, Lists, etc.) will be automatically serializable using [json4s](http://json4s.org)

edcote, 2017-12-27:
Is the existing YAML infrastructure going to stay?

jackkoenig, 2017-12-27:
I was planning to deprecate it and then remove sometime next year. The primary issue with it is that the current implementation and library cannot easily distinguish between different Annotation types that have the same fields, json4s provides nice automatic type tagging for dealing with that.

edcote, 2017-12-27:
Thanks, Jack.  I'll monitor the release notes.  Should not be a problem for me to make the transition.

colinschmidt, 2018-06-22:
This file should be gone I assume? 
https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/passes/memlib/YamlUtils.scala

seldridge, 2018-06-22:
Looks like this is not ready to be closed... Those `YamlFileReader` and `YamlFileWriter` are still being used. `Wiring` is entirely moved to using annotations (and resulting JSON), though.

sequencer, 2019-06-14:
Is memlib ready to be remove from next version of firrtl?
I tried to port firrtl to Scala 2.13
`moultingyaml` seem to be abandoned, and it's too old to port to Scala 2.13, which will become the only barrier for Scala 2.13 porting.

jackkoenig, 2019-06-21:
1.2 will be out soon, we should fix this and bump to Scala 2.13 in 1.3