chipsalliance/firrtl #457: How does BlackBoxResourceHelper work?

shunshou, 2017-02-22:
@chick 

It seems to be doing the right thing but -- if I add a BlackBoxResource in Firrtl after the pass is run, will it figure out that there's a new BlackBox annotation that needs to be handled and rerun it? It seems like it does, but I don't get the mechanism for how it works...

shunshou, 2017-02-22:
Starting ***LowFirrtlOptimization***
Starting Remove ValidIfs
Finished Remove ValidIfs
Remove ValidIfs took 0.1 ms

Starting Constant Propagation
Finished Constant Propagation
Constant Propagation took 0.2 ms

Starting Pad Widths
Finished Pad Widths
Pad Widths took 0.3 ms

Starting Constant Propagation
Finished Constant Propagation
Constant Propagation took 0.4 ms

Starting Legalize
Finished Legalize
Legalize took 0.2 ms

Starting Verilog Memory Delays
Finished Verilog Memory Delays
Verilog Memory Delays took 0.3 ms

Starting Constant Propagation
Finished Constant Propagation
Constant Propagation took 0.2 ms

Starting Split Expressions
Finished Split Expressions
Split Expressions took 0.2 ms

Starting Common Subexpression Elimination
Finished Common Subexpression Elimination
Common Subexpression Elimination took 0.3 ms

Starting Dead Code Elimination
Finished Dead Code Elimination
Dead Code Elimination took 0.2 ms

Finished ***LowFirrtlOptimization***
***LowFirrtlOptimization*** took 3.8 ms

Starting ***BlackBoxSourceHelper***
Finished ***BlackBoxSourceHelper***
***BlackBoxSourceHelper*** took 0.5 ms

Starting ***AddIOPadsTransform***
Starting Legalize
Finished Legalize
Legalize took 0.3 ms

Starting Resolve Genders
Finished Resolve Genders
Resolve Genders took 0.2 ms

Starting Add Padframe
Finished Add Padframe
Add Padframe took 1.0 ms

Starting Remove Empty Statements
Finished Remove Empty Statements
Remove Empty Statements took 0.1 ms

Starting Check Initialization
Finished Check Initialization
Check Initialization took 0.3 ms

Starting Infer Types
Finished Infer Types
Infer Types took 2.2 ms

Starting Uniquify Identifiers
Finished Uniquify Identifiers
Uniquify Identifiers took 4.0 ms

Starting Resolve Kinds
Finished Resolve Kinds
Resolve Kinds took 1.2 ms

Starting Resolve Genders
Finished Resolve Genders
Resolve Genders took 0.6 ms

Finished ***AddIOPadsTransform***
***AddIOPadsTransform*** took 70.7 ms

Starting ***BlackBoxSourceHelper***
Finished ***BlackBoxSourceHelper***
***BlackBoxSourceHelper*** took 1.1 ms

Starting Add temporary nodes with verilog widths for modulus
Finished Add temporary nodes with verilog widths for modulus
Add temporary nodes with verilog widths for modulus took 0.4 ms

Starting Verilog Wrap
Finished Verilog Wrap
Verilog Wrap took 0.5 ms

Starting Verilog Rename
Finished Verilog Rename
Verilog Rename took 0.9 ms

Starting Verilog Prep
Finished Verilog Prep
Verilog Prep took 2.5 ms

chick, 2017-02-23:
@shunshou This is a mystery to me too, seems like nice behavior but I don't know how it happens.  I'll take a look and see what's going on

shunshou, 2017-02-27:
Ok, so you added BlackBoxSourceHelper to compiler.transforms, which gets auto-run first. I had a black box in my Module, so it ran it a second time (after add pads, b/c that was the order in which things were added). It doesn't seem smart enough to add transforms based off of new annotations.

shunshou, 2017-02-27:
I created a new instance of the BB transform within my addPads transform and executed it. the *.f file (with all the Verilog file names) gets overwritten by a second run of the transform... It doesn't seem like the *.f file is used anywhere, but maybe it should be appending to instead of overwriting? (and maybe the directory should be wiped at the start of each compile?)

colinschmidt, 2017-02-27:
I think it makes sense to append rather than overwrite, while removing the file at the start of each compile.

azidar, 2017-12-15:
Should be revisited with artifacts.