chipsalliance/firrtl #469: A proposal for better generated names

ben-k, 2017-02-27:
Sometimes I have to grep through the generated Verilog for a wire/reg/etc. named `_GEN*`.

This process would be a bit more user friendly if the Firrtl-generated names were distinguished by type.  So wires could be named `_GEN_W*`, registers `_GEN_R*`, and so forth.

I imagine (hope) this would not be a burdensome change, and it would definitely improve readability.

See also my related request: ucb-bar/chisel3#535

jackkoenig, 2017-02-27:
It's a lot more burdensome here than in Chisel (and thus will probably happen in stages rather than one fell swoop). Firrtl also almost exclusively generates wires and nodes which in their use here are almost always identical. Would it perhaps be better to start trying to have things named based on the name of whatever thing leads to their creation? If a vec named `myROM` creates a bunch of nodes they could be named something like `_GEN_myROM_1`?

ben-k, 2017-02-27:
That would be even better!

jackkoenig, 2017-02-27:
Thoughts @azidar?

azidar, 2017-03-16:
I have implemented something like this before - the problem was if a _GEN_ helps generate another _GEN_, then you get names like _GEN_94_GEN_41_myRom_1 etc. I think its certainly worth spending some time on, but the solution isn't simple.