chipsalliance/firrtl #475: Generic Annotation

shunshou, 2017-03-02:
I think we talked about this earlier, but I'm just going to file an issue to track -- I'd like to have some generic annotation that applies to the entire circuit. That way, I don't need to use my BlackBoxSourceHelper hack to get stuff like targetDir name. (Or input config file location).

jackkoenig, 2017-03-02:
Thanks for tracking this. I actually have a related thing in my PR here: https://github.com/ucb-bar/firrtl/blob/9a6a2a05abca97a4ec178112083a036712146d03/src/main/scala/firrtl/annotations/Named.scala#L19

But that only solves the "Named" part. I think the way of dealing with the target transform is to use `classOf[Transform]` (instead of the class of a specific transform). However, we need to change [`getMyAnnotations`](https://github.com/ucb-bar/firrtl/blob/master/src/main/scala/firrtl/Compiler.scala#L106) to return these annotations as well.

jackkoenig, 2018-01-03:
Will be fixed by annotations refactor

commit c7eb1570 (2018-02-28): Refactor Annotations (#721)

- Old Annotation renamed to deprecated LegacyAnnotation
- Annotation is now a trait that can be extended
- New JsonProtocol for Annotation [de]serialization
- Replace AnnotationMap with AnnotationSeq
- Deprecate Transform.getMyAnnotations
- Update Transforms
- Turn on deprecation warnings
- Remove deprecated Driver.compile
- Make AnnotationTests abstract with Legacy and Json subclasses
- Add functionality to convert LegacyAnnotations of built-in annos
This will give a noisy warning and is more of a best effort than a
robust solution.

Fixes #475 Closes #609
closed 2018-02-28 by jackkoenig