chipsalliance/firrtl #489: Internal Error: From dsptools ResetInverterSpec

chick, 2017-03-11:
In ucb-bar/dsptools on branch signBit-div2-mul2
The test file ResetNSpec crashes with 
```
[info] [0.004] Elaborating design...
[info] [0.085] Done elaborating.

Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
firrtl.FIRRTLException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
	at firrtl.Utils$.error(Utils.scala:363)
	at firrtl.Utils$.throwInternalError(Utils.scala:132)
	at firrtl.Driver$.execute(Driver.scala:183)
	at chisel3.Driver$.execute(Driver.scala:161)
	at dsptools.ResetNSpec$$anonfun$1.apply$mcV$sp(ResetInverterSpec.scala:26)
	at dsptools.ResetNSpec$$anonfun$1.apply(ResetInverterSpec.scala:22)
	at dsptools.ResetNSpec$$anonfun$1.apply(ResetInverterSpec.scala:22)
	at org.scalatest.Transformer$$anonfun$apply$1.apply$mcV$sp(Transformer.scala:22)
	at org.scalatest.OutcomeOf$class.outcomeOf(OutcomeOf.scala:85)
	at org.scalatest.OutcomeOf$.outcomeOf(OutcomeOf.scala:104)
	at org.scalatest.Transformer.apply(Transformer.scala:22)
	at org.scalatest.Transformer.apply(Transformer.scala:20)
	at org.scalatest.FreeSpecLike$$anon$1.apply(FreeSpecLike.scala:373)
	at org.scalatest.Suite$class.withFixture(Suite.scala:1122)
	at org.scalatest.FreeSpec.withFixture(FreeSpec.scala:1740)
	at org.scalatest.FreeSpecLike$class.invokeWithFixture$1(FreeSpecLike.scala:370)
	at org.scalatest.FreeSpecLike$$anonfun$runTest$1.apply(FreeSpecLike.scala:382)
	at org.scalatest.FreeSpecLike$$anonfun$runTest$1.apply(FreeSpecLike.scala:382)
	at org.scalatest.SuperEngine.runTestImpl(Engine.scala:306)
	at org.scalatest.FreeSpecLike$class.runTest(FreeSpecLike.scala:382)
	at org.scalatest.FreeSpec.runTest(FreeSpec.scala:1740)
	at org.scalatest.FreeSpecLike$$anonfun$runTests$1.apply(FreeSpecLike.scala:441)
	at org.scalatest.FreeSpecLike$$anonfun$runTests$1.apply(FreeSpecLike.scala:441)
	at org.scalatest.SuperEngine$$anonfun$traverseSubNodes$1$1.apply(Engine.scala:413)
	at org.scalatest.SuperEngine$$anonfun$traverseSubNodes$1$1.apply(Engine.scala:401)
	at scala.collection.immutable.List.foreach(List.scala:381)
	at org.scalatest.SuperEngine.traverseSubNodes$1(Engine.scala:401)
	at org.scalatest.SuperEngine.org$scalatest$SuperEngine$$runTestsInBranch(Engine.scala:396)
	at org.scalatest.SuperEngine.runTestsImpl(Engine.scala:483)
	at org.scalatest.FreeSpecLike$class.runTests(FreeSpecLike.scala:441)
	at org.scalatest.FreeSpec.runTests(FreeSpec.scala:1740)
	at org.scalatest.Suite$class.run(Suite.scala:1424)
	at org.scalatest.FreeSpec.org$scalatest$FreeSpecLike$$super$run(FreeSpec.scala:1740)
	at org.scalatest.FreeSpecLike$$anonfun$run$1.apply(FreeSpecLike.scala:486)
	at org.scalatest.FreeSpecLike$$anonfun$run$1.apply(FreeSpecLike.scala:486)
	at org.scalatest.SuperEngine.runImpl(Engine.scala:545)
	at org.scalatest.FreeSpecLike$class.run(FreeSpecLike.scala:486)
	at org.scalatest.FreeSpec.run(FreeSpec.scala:1740)
	at org.scalatest.tools.SuiteRunner.run(SuiteRunner.scala:55)
	at org.scalatest.tools.Runner$$anonfun$doRunRunRunDaDoRunRun$3.apply(Runner.scala:2563)
	at org.scalatest.tools.Runner$$anonfun$doRunRunRunDaDoRunRun$3.apply(Runner.scala:2557)
	at scala.collection.immutable.List.foreach(List.scala:381)
	at org.scalatest.tools.Runner$.doRunRunRunDaDoRunRun(Runner.scala:2557)
	at org.scalatest.tools.Runner$$anonfun$runOptionallyWithPassFailReporter$2.apply(Runner.scala:1044)
	at org.scalatest.tools.Runner$$anonfun$runOptionallyWithPassFailReporter$2.apply(Runner.scala:1043)
	at org.scalatest.tools.Runner$.withClassLoaderAndDispatchReporter(Runner.scala:2722)
	at org.scalatest.tools.Runner$.runOptionallyWithPassFailReporter(Runner.scala:1043)
	at org.scalatest.tools.Runner$.run(Runner.scala:883)
	at org.scalatest.tools.Runner.run(Runner.scala)
	at org.jetbrains.plugins.scala.testingSupport.scalaTest.ScalaTestRunner.runScalaTest2(ScalaTestRunner.java:138)
	at org.jetbrains.plugins.scala.testingSupport.scalaTest.ScalaTestRunner.main(ScalaTestRunner.java:28)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:483)
	at com.intellij.rt.execution.application.AppMain.main(AppMain.java:147)
```

azidar, 2017-03-13:
Can you git gist the firrtl file?

chick, 2017-03-13:
It's in branch [here -- ResetInverterSpec.scala](https://github.com/ucb-bar/dsptools/blob/signBit-div2-mul2/src/test/scala/dsptools/ResetInverterSpec.scala)

chick, 2017-03-13:
@azidar It's small so here it is
```
;buildInfoPackage: chisel3, version: 3.1-SNAPSHOT, scalaVersion: 2.11.7, sbtVersion: 0.13.12, builtAtString: 2017-03-13 18:53:36.116, builtAtMillis: 1489431216116
circuit ExampleModuleNeedsResetInverted :
  module ExampleModuleNeedsResetInverted :
    input clock : Clock
    input reset : UInt<1>
    output io : {out : UInt<32>}

    io is invalid
    io is invalid
    reg r : UInt<32>, clock with : (reset => (reset, UInt<32>("h00"))) @[ResetInverterSpec.scala 15:18]
```

closed 2017-03-17 by chick