chipsalliance/firrtl #491: FIRRTL Internal Error when generating Verilog with firrtl.Driver.compile()

edwardcwang, 2017-03-12:
FIRRTL revision e571ef88f7f69b2374fa9ba86e219523645213c6

Code:
https://github.com/ucb-bar/barstools/blob/f1c437f83079160fddcca8f6a102f605dab5ea31/tapeout/src/main/scala/transforms/GenerateTop.scala

[MicroDemoTop.fir](https://gist.github.com/edwardcwang/e99f90f734083fa08e0ac1a9bf3a1392) (if needed)

'run-main barstools.tapeout.transforms.GenerateTop -i MicroDemoTop.fir -o MicroDemoTop.v --syn-top MicroDemoTop --harness-top MicroDemoTop'

```
[error] (run-main-0) firrtl.FIRRTLException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
firrtl.FIRRTLException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
	at firrtl.Utils$.error(Utils.scala:363)
	at firrtl.Utils$.throwInternalError(Utils.scala:132)
	at firrtl.Compiler$class.compile(Compiler.scala:276)
	at firrtl.VerilogCompiler.compile(LoweringCompilers.scala:137)
	at firrtl.Driver$.compile(Driver.scala:56)
	at barstools.tapeout.transforms.GenerateTop$.delayedEndpoint$barstools$tapeout$transforms$GenerateTop$1(GenerateTop.scala:53)
	at barstools.tapeout.transforms.GenerateTop$delayedInit$body.apply(GenerateTop.scala:10)
	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
	at scala.App$$anonfun$main$1.apply(App.scala:76)
	at scala.App$$anonfun$main$1.apply(App.scala:76)
	at scala.collection.immutable.List.foreach(List.scala:381)
	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
	at scala.App$class.main(App.scala:76)
	at barstools.tapeout.transforms.GenerateTop$.main(GenerateTop.scala:10)
	at barstools.tapeout.transforms.GenerateTop.main(GenerateTop.scala)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:57)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:606)

```

edwardcwang, 2017-03-12:
@azidar It seems to have broken in b5ef5b876d4f4ad4a17bc81362b2264970272d63; would you be able to take a look?

azidar, 2017-03-15:
I couldn't duplicate your error - are you sure that the version of firrtl you think you're using is the one you've actually publish-local?

Try deleting your .ivy2 cache, and publishing the current master branch of firrtl, and let me know if you still see that error.

azidar, 2017-03-15:
What happened is the API changed for transforms, so the transforms in the old version of barstools is broken when using the new Firrtl API. If you checkout the latest version of barstools, you'll see that it works.

I will change the error message so it doesn't throw an internal error, but a more reasonable error message.

edwardcwang, 2017-03-15:
Changing the error message to be more reasonable (and may help other developers as well) sounds great, thanks!

azidar, 2017-03-15:
Note though, please use the latest version of barstools - otherwise, none of the passes in barstools are going to work. Sure thing :)

PR #494 (2017-03-22): Throw different error message for missing emitanno
Fix #491
closed 2017-03-15 by azidar