chipsalliance/firrtl #496: Make read-write conflicts actually undefined in verilog emission

azidar, 2017-03-16:
Right now, we generate a vector of registers, so the behavior is, while still legal, can be misleading for people.

@shunshou any additional comments?

shunshou, 2017-03-16:
That's it. Just a scenario where this became a problem: 

I have a design that allows reads/writes to happen simultaneously (sometimes to the same address). I wrote some logic that theoretically passed the written value to data out when the address conflict occurred, but I actually screwed that up. It's just that w/ Xilinx + Chisel sims, I never caught the problem, but then the behavior with real SRAMs was undefined and tests that were passing ended up failing. Took some time to figure out inconsistencies. Would definitely prefer to have caught it earlier.

ben-k, 2017-03-20:
I'm trying to figure out if this is related to ucb-bar/rocket-chip/issues/548 - still not certain, but I wanted to at least make the connection in case anyone from either side has something to add to the other issue.

azidar, 2017-12-15:
subsumed by #635

closed 2017-12-15 by azidar