chipsalliance/firrtl #498: Firrtl Match Error Shlw

shunshou, 2017-03-20:
I can't get my design through Firrtl :(

Looks like the width padding is adding shlw PrimOps that aren't matched properly? I'm super confused as to why this is a problem for me but hasn't been seen by anyone else.

```
[info] - should read in ADC inputs *** FAILED ***
[info]   scala.MatchError: shlw (of class firrtl.Shlw$)
[info]   at firrtl.PrimOps$.set_primop_type(PrimOps.scala:129)
[info]   at firrtl.passes.InferTypes$.firrtl$passes$InferTypes$$infer_types_e$1(InferTypes.scala:32)
[info]   at firrtl.passes.InferTypes$$anonfun$firrtl$passes$InferTypes$$infer_types_s$1$7.apply(InferTypes.scala:60)
[info]   at firrtl.passes.InferTypes$$anonfun$firrtl$passes$InferTypes$$infer_types_s$1$7.apply(InferTypes.scala:60)
[info]   at firrtl.ir.Connect.mapExpr(IR.scala:240)
[info]   at firrtl.Mappers$StmtMagnet$$anon$2.map(Mappers.scala:19)
[info]   at firrtl.Mappers$StmtMap$.map$extension(Mappers.scala:30)
[info]   at firrtl.passes.InferTypes$.firrtl$passes$InferTypes$$infer_types_s$1(InferTypes.scala:60)
[info]   at firrtl.passes.InferTypes$$anonfun$firrtl$passes$InferTypes$$infer_types_s$1$5.apply(InferTypes.scala:60)
[info]   at firrtl.passes.InferTypes$$anonfun$firrtl$passes$InferTypes$$infer_types_s$1$5.apply(InferTypes.scala:60)
```

shunshou, 2017-03-20:
After further debugging:

My ClkSrcTransform (LowForm) runs an InferTypes pass before starting. Does that mean InferTypes doesn't work in LowFirrtl? I never caught it before, b/c I wasn't using it with designs that required padding widths...

```
Form: LowForm
======== Finished Transform MiddleFirrtlToLowFirrtl ========

======== Starting Transform LowFirrtlOptimization ========
-------- Starting Pass Remove ValidIfs --------
Time: 19.4 ms
-------- Finished Pass Remove ValidIfs --------
-------- Starting Pass Constant Propagation --------
Time: 48.1 ms
-------- Finished Pass Constant Propagation --------
-------- Starting Pass Pad Widths --------
Time: 28.5 ms
-------- Finished Pass Pad Widths --------
-------- Starting Pass Constant Propagation --------
Time: 18.3 ms
-------- Finished Pass Constant Propagation --------
-------- Starting Pass Legalize --------
Time: 13.5 ms
-------- Finished Pass Legalize --------
-------- Starting Pass Verilog Memory Delays --------
Time: 35.5 ms
-------- Finished Pass Verilog Memory Delays --------
-------- Starting Pass Constant Propagation --------
Time: 3.4 ms
-------- Finished Pass Constant Propagation --------
-------- Starting Pass Split Expressions --------
Time: 18.2 ms
-------- Finished Pass Split Expressions --------
-------- Starting Pass Common Subexpression Elimination --------
Time: 96.0 ms
-------- Finished Pass Common Subexpression Elimination --------
-------- Starting Pass Dead Code Elimination --------
Time: 28.0 ms
-------- Finished Pass Dead Code Elimination --------
Time: 316.6 ms
Form: LowForm
======== Finished Transform LowFirrtlOptimization ========

======== Starting Transform BlackBoxSourceHelper ========
Time: 35.8 ms
Form: LowForm
======== Finished Transform BlackBoxSourceHelper ========

======== Starting Transform BlackBoxSourceHelper ========
Time: 17.9 ms
Form: LowForm
======== Finished Transform BlackBoxSourceHelper ========

======== Starting Transform ClkSrcTransform ========
-------- Starting Pass Infer Types --------
```

jackkoenig, 2017-03-20:
This has been a problem that creeps up off and on. There are certain Primops that aren't really legal Firrtl but are instead just to simplify conversion to Verilog. I tried to move all of those passes to only be run in the actual Verilog emitter so that our transforms never return illegal Firrtl, but it might have been a lot of work to pull it out in this case (or I might just have missed it).

donggyukim, 2017-07-14:
I got this error after running `LowFirrtlOptimation`, custom transforms, and a whole compiler again. Is there any thoughts/progress on this?

jackkoenig, 2017-07-14:
Related discussion here: https://github.com/freechipsproject/firrtl/pull/500.

PadWidths needs to not introduce shlw and dshlw, those should only be introduced in the Verilog emitter

Thoughts @azidar

azidar, 2017-12-15:
Refactor necessary for LowFirrtlOptimization and VerilogEmitter. This includes having an InlineExpression transform, and a check to ensure the circuit can be legally emitted to Verilog.

chick, 2018-12-17:
Should be fixed by custom IR node stuff

albert-magyar, 2020-03-02:
Closed, absorbed into #1410.

closed 2020-03-02 by albert-magyar