chipsalliance/firrtl #499: Using High Firrtl Compiler error

shunshou, 2017-03-20:
If I set compiler to high Firrtl, I get this error:

```
fo] FFASTTopBuildSpec:
[info] FFASTTopBuild
[info] - should read in ADC inputs *** FAILED ***
[info]   java.lang.AssertionError: assertion failed: No transform in List(firrtl.ChirrtlToHighFirrtl@7e52b3e6) has outputForm LowForm as required by firrtl.transforms.BlackBoxSourceHelper@67a101d4
[info]   at scala.Predef$.assert(Predef.scala:165)
[info]   at firrtl.CompilerUtils$$anonfun$mergeTransforms$1.apply(Compiler.scala:228)
[info]   at firrtl.CompilerUtils$$anonfun$mergeTransforms$1.apply(Compiler.scala:226)
[info]   at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
[info]   at scala.collection.immutable.List.foldLeft(List.scala:84)
[info]   at firrtl.CompilerUtils$.mergeTransforms(Compiler.scala:226)
[info]   at firrtl.Compiler$class.compile(Compiler.scala:314)
[info]   at firrtl.HighFirrtlCompiler.compile(LoweringCompilers.scala:119)
[info]   at firrtl.Driver$.execute(Driver.scala:167)
[info]   at chisel3.Driver$.execute(Driver.scala:161)
```

Seems like an implementation bug. I assume you strip all of the LowForm transforms away if the high firrtl compiler is used? Idk.

jackkoenig, 2017-03-20:
Are you passing a custom transform that is supposed to run on Low Firrtl? The error message could be better but I'm pretty sure that's what's going on.

It does seem like there is a reasonable use case of people trying to look at the High Firrtl before their Low Firrtl custom pass is run. Need to think about how to deal with this.

shunshou, 2017-03-20:
Sorry if I wasn't clear -- that's exactly what I was trying to do.

jackkoenig, 2017-03-20:
What do you think the behavior should be? Just don't run custom transforms that are supposed to be run on a lower form than the compiler uses?

I originally chose this behavior because I worried about people invoking Firrtl with their custom transform and then being surprised that it doesn't get run. Do you think this is a problem? (Obviously the current behavior of erroring needs to change, especially since annotations add custom transforms which could trip this.)

shunshou, 2017-03-20:
Yea I was using the transform via Annotations. I think custom transforms that should be run afterwards should just not be run. If you "worry," I think it's sufficient just to give a list of custom transforms that weren't run due to using a different compiler...

jackkoenig, 2017-03-20:
Yeah I was thinking about having compilers print a list of transforms they are going to run (with order) and custom transforms not run to log-level warn or info.

colinschmidt, 2017-03-20:
I think a list of passes being run would be very helpful at some log-level.

albert-magyar, 2020-03-09:
This issue is rendered obsolete by a combination of #920 and #1357

closed 2020-03-09 by albert-magyar