chipsalliance/firrtl #501: Reg(Vec()) generation is sub-optimal, includes dead code.

ccelio, 2017-03-21:
**TL;DR:** lots of dead code shows up in Reg(Vec()) Verilog output. Exacerbated by Bundles. Probably affecting compile time. Example provided.

------
I just updated BOOM to a Feb 10th commit of rocket-chip (last updated in ~November). I've noticed a significant increase in Verilog code size, a huge increase in compile time (4 minutes to compile Verilator to 12+ minutes now), and a corresponding slowdown in simulation speed. My compile/run/debug loop has gone from ~6 minutes to 25-30 minutes.

I realize this is a mash of rocket-chip, chisel3, and FIRRTL changes, but bear with me.

I am using the following commits (admittedly ~2 month old, but i don't see any changes to Vec behavior): 
firrtl- 568f25b221884eeb0db362c902c933f734c7e47e
chisel3- 3ef63639284b2b56f415e1540c58d85d88c360db
rocket-chip- 5fc44bbcda9c5705fcec48a86c70047f283095a8

(Also, I tried the latest chisel/firttl, but it just crashes out of JavaHeapSpace on `ZeroWidth` pass).

-------
Here's the size of generated output:
````
15M	rocketchip.BOOMConfig.fir
30M	rocketchip.BOOMConfig.v

12M	rocketchip.DefaultConfig.fir
11M	rocketchip.DefaultConfig.v

18M	rocketchip.MegaBoomConfig.fir
80M	rocketchip.MegaBoomConfig.v
````

(A Nov 6th BOOMConfig was only 7Mb of firrtl code.)

As evident from BOOM, a large part of the disparity from `.fir` to `.v` is Vec generation, and the largest Modules as measured by LOC have a lot of Vecs in them.

------

**An Example:**

````scala
class MyDebugModule extends Module {                                 
   val io = new Bundle {                              
      val in = Bool(INPUT)        
      val ridx = UInt(INPUT, 2)   
      val widx = UInt(INPUT, 2)   
      val out = Bool(OUTPUT)      
   }                              
   val myvec = Reg(Vec(4, Bool()))

   myvec(io.widx) := io.in        
   io.out := myvec(io.ridx)       
}                                 
````

Firrtl:
````scala
   module MyDebugModule :                                                                     
     input clock : Clock                                                                      
     input reset : UInt<1>                                                                    
     output io : {flip in : UInt<1>, flip ridx : UInt<2>, flip widx : UInt<2>, out : UInt<1>} 
                                                                                              
     io is invalid                                                                            
     io is invalid                                                                            
     reg myvec : UInt<1>[4], clock @[core.scala 60:19]                                        
     myvec[io.widx] <= io.in @[core.scala 62:19]                                              
     io.out <= myvec[io.ridx] @[core.scala 63:11]                                                             
````

Verilog:

````verilog
module MyDebugModule(                                  
  input   clock,                                       
  input   reset,                                       
  input   io_in,                                       
  input  [1:0] io_ridx,                                
  input  [1:0] io_widx,                                
  output  io_out                                       
);                                                     
  reg  myvec_0;                                        
  reg [31:0] _GEN_9;                                   
  reg  myvec_1;                                        
  reg [31:0] _GEN_10;                                  
  reg  myvec_2;                                        
  reg [31:0] _GEN_11;                                  
  reg  myvec_3;                                        
  reg [31:0] _GEN_12;                                  
  wire  _GEN_0;                                        
  wire  _GEN_2;                                        
  wire  _GEN_3;                                        
  wire  _GEN_4;                                        
  wire  _GEN_5;                                        
  wire  _GEN_1;                                        
  wire  _GEN_6;                                        
  wire  _GEN_7;                                        
  wire  _GEN_8;                                        
  assign io_out = _GEN_1;                              
  assign _GEN_0 = io_in;                               
  assign _GEN_2 = 2'h0 == io_widx ? _GEN_0 : myvec_0;  
  assign _GEN_3 = 2'h1 == io_widx ? _GEN_0 : myvec_1;  
  assign _GEN_4 = 2'h2 == io_widx ? _GEN_0 : myvec_2;  
  assign _GEN_5 = 2'h3 == io_widx ? _GEN_0 : myvec_3;  
  assign _GEN_1 = _GEN_8;                              
  assign _GEN_6 = 2'h1 == io_ridx ? myvec_1 : myvec_0; 
  assign _GEN_7 = 2'h2 == io_ridx ? myvec_2 : _GEN_6;  
  assign _GEN_8 = 2'h3 == io_ridx ? myvec_3 : _GEN_7;  
`ifdef RANDOMIZE                                       
  integer initvar;                                     
  initial begin                                        
    `ifndef verilator                                  
      #0.002 begin end                                 
    `endif                                             
  `ifdef RANDOMIZE_REG_INIT                            
  _GEN_9 = {1{$random}};                               
  myvec_0 = _GEN_9[0:0];                               
  `endif                                               
  `ifdef RANDOMIZE_REG_INIT                            
  _GEN_10 = {1{$random}};                              
  myvec_1 = _GEN_10[0:0];                              
  `endif                                               
  `ifdef RANDOMIZE_REG_INIT                            
  _GEN_11 = {1{$random}};                              
  myvec_2 = _GEN_11[0:0];                              
  `endif                                               
  `ifdef RANDOMIZE_REG_INIT                            
  _GEN_12 = {1{$random}};                              
  myvec_3 = _GEN_12[0:0];                              
  `endif                                               
  end                                                  
`endif                                                 
   always @(posedge clock) begi
     if (2'h0 == io_widx) begin
       myvec_0 <= _GEN_0;      
     end                       
     if (2'h1 == io_widx) begin
       myvec_1 <= _GEN_0;      
     end                       
     if (2'h2 == io_widx) begin
       myvec_2 <= _GEN_0;      
     end                       
     if (2'h3 == io_widx) begin
       myvec_3 <= _GEN_0;      
     end                       
   end                         
 endmodule                     
````

A couple of things:
* The random initialization code is 5 lines for every entry in the Vec. 
* _GEN_{2,3,4,5} are deadcode.  Although it won't be synthesized, I assume it will still affect compile time.
* This is an increase from 10 lines of code to 75 loc. 
* A muxtree is used to read out data, although that's always been the case.
* This issues only get worse if you use a Reg(Vec(Bundle())). Each element is a new Vec (for good reason), but unconnected Bundle elements are still emitted (random initialization, and the read muxtree is emitted). Boom has lots of Vecs of Bundles with unused fields, so I'm hit especially hard.

aswaterman, 2017-03-21:
Last I checked, dead code elimination doesn't delete wires/regs, and could stand to be improved.

jackkoenig, 2017-03-21:
```
(Also, I tried the latest chisel/firttl, but it just crashes out of JavaHeapSpace on ZeroWidth pass).
```

Sounds like a separate bug. Would you mind filing a separate issue?

azidar, 2017-03-22:
see #506, it probably fixes the DCE problem.

azidar, 2017-12-15:
Fixed.

closed 2017-12-15 by azidar