chipsalliance/firrtl #502: Java heap space OutOfMemoryError in ZeroWidth pass.

ccelio, 2017-03-21:
**Output:**
````

[info] [11.192] Done elaborating.
[success] Total time: 121 s, completed Mar 21, 2017 2:02:53 PM
mkdir -p /scratch/celio/firrtl-chip-deleteme/emulator/generated-src/
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp /scratch/celio/firrtl-chip-deleteme/firrtl/utils/bin/firrtl.jar firrtl.Driver -i /scratch/celio/firrtl-chip-deleteme/emulator/generated-src/rocketchip.BOOMConfig.fir -o /scratch/celio/firrtl-chip-deleteme/emulator/generated-src/rocketchip.BOOMConfig.v -X verilog
Exception in thread "main" java.lang.OutOfMemoryError: Java heap space
	at scala.collection.mutable.ResizableArray$class.ensureSize(ResizableArray.scala:103)
	at scala.collection.mutable.ArrayBuffer.ensureSize(ArrayBuffer.scala:48)
	at scala.collection.mutable.ArrayBuffer.$plus$eq(ArrayBuffer.scala:84)
	at scala.collection.mutable.ArrayBuffer.$plus$eq(ArrayBuffer.scala:48)
	at scala.collection.TraversableLike$$anonfun$collect$1.apply(TraversableLike.scala:282)
	at scala.collection.TraversableLike$$anonfun$collect$1.apply(TraversableLike.scala:282)
	at scala.PartialFunction$$anonfun$runWith$1.apply(PartialFunction.scala:142)
	at scala.PartialFunction$$anonfun$runWith$1.apply(PartialFunction.scala:140)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at scala.collection.TraversableLike$class.collect(TraversableLike.scala:282)
	at scala.collection.AbstractTraversable.collect(Traversable.scala:104)
	at firrtl.passes.ZeroWidth$.firrtl$passes$ZeroWidth$$removeZero(ZeroWidth.scala:19)
	at firrtl.passes.ZeroWidth$.firrtl$passes$ZeroWidth$$removeZero(ZeroWidth.scala:25)
	at firrtl.passes.ZeroWidth$.firrtl$passes$ZeroWidth$$onExp(ZeroWidth.scala:28)
	at firrtl.passes.ZeroWidth$$anonfun$firrtl$passes$ZeroWidth$$onExp$3.apply(ZeroWidth.scala:36)
	at firrtl.passes.ZeroWidth$$anonfun$firrtl$passes$ZeroWidth$$onExp$3.apply(ZeroWidth.scala:36)
	at firrtl.WSubIndex.mapExpr(WIR.scala:51)
	at firrtl.Mappers$ExprMagnet$$anon$5.map(Mappers.scala:39)
	at firrtl.Mappers$ExprMap$.map$extension(Mappers.scala:49)
	at firrtl.passes.ZeroWidth$.firrtl$passes$ZeroWidth$$onExp(ZeroWidth.scala:36)
	at firrtl.passes.ZeroWidth$$anonfun$firrtl$passes$ZeroWidth$$onExp$3.apply(ZeroWidth.scala:36)
	at firrtl.passes.ZeroWidth$$anonfun$firrtl$passes$ZeroWidth$$onExp$3.apply(ZeroWidth.scala:36)
	at firrtl.WSubField.mapExpr(WIR.scala:42)
	at firrtl.Mappers$ExprMagnet$$anon$5.map(Mappers.scala:39)
	at firrtl.Mappers$ExprMap$.map$extension(Mappers.scala:49)
	at firrtl.passes.ZeroWidth$.firrtl$passes$ZeroWidth$$onExp(ZeroWidth.scala:36)
	at firrtl.passes.ZeroWidth$$anonfun$firrtl$passes$ZeroWidth$$onExp$3.apply(ZeroWidth.scala:36)
	at firrtl.passes.ZeroWidth$$anonfun$firrtl$passes$ZeroWidth$$onExp$3.apply(ZeroWidth.scala:36)
	at firrtl.ir.Mux.mapExpr(IR.scala:90)
	at firrtl.Mappers$ExprMagnet$$anon$5.map(Mappers.scala:39)
	at firrtl.Mappers$ExprMap$.map$extension(Mappers.scala:49)
make: *** [/scratch/celio/firrtl-chip-deleteme/emulator/generated-src/rocketchip.BOOMConfig.v] Error 1

real	7m13.858s
user	71m8.369s
sys	0m39.988s

````

-----
**To reproduce:**
````bash
git clone https://github.com/ucb-bar/rocket-chip.git firrtl-chip
cd firrtl-chip
git checkout boom-devel 
git submodule update --init
cd firrtl; git checkout master; git pull
cd ../emulator;
time make
````

The full hash of rocket-chip is 39ec5a185bd82068b93a932c9cbc98f1b68b84ce.
The firrtl hash is 13fc27f35e85026c002e644b61c32268bd258d78.
The chisel hash is 3ef63639284b2b56f415e1540c58d85d88c360db.

azidar, 2017-03-21:
While I appreciate the reproduction steps, its easier for me if you could just provide /scratch/celio/firrtl-chip-deleteme/emulator/generated-src/rocketchip.BOOMConfig.fir. Thanks!

ccelio, 2017-03-21:
Alright, it's at /nscratch/celio/for_adam/*.2017.fir

azidar, 2017-03-21:
If I run it with:
```java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp firrtl.jar -i...```
I get
```
Exception in thread "main" java.lang.OutOfMemoryError: GC overhead limit exceeded
        at firrtl.passes.ZeroWidth$$anonfun$3.apply(ZeroWidth.scala:19)
        at firrtl.passes.ZeroWidth$$anonfun$3.apply(ZeroWidth.scala:19)
        at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
        at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
        at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
        at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
        at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
        at scala.collection.AbstractTraversable.map(Traversable.scala:104)
        at firrtl.passes.ZeroWidth$.firrtl$passes$ZeroWidth$$removeZero(ZeroWidth.scala:19)
        at firrtl.passes.ZeroWidth$$anonfun$3.apply(ZeroWidth.scala:19)
        at firrtl.passes.ZeroWidth$$anonfun$3.apply(ZeroWidth.scala:19)
        at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
        at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
        at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
        at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
        at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
        at scala.collection.AbstractTraversable.map(Traversable.scala:104)
        at firrtl.passes.ZeroWidth$.firrtl$passes$ZeroWidth$$removeZero(ZeroWidth.scala:19)
        at firrtl.passes.ZeroWidth$$anonfun$3.apply(ZeroWidth.scala:19)
        at firrtl.passes.ZeroWidth$$anonfun$3.apply(ZeroWidth.scala:19)
        at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
        at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
        at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
        at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
        at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
        at scala.collection.AbstractTraversable.map(Traversable.scala:104)
        at firrtl.passes.ZeroWidth$.firrtl$passes$ZeroWidth$$removeZero(ZeroWidth.scala:19)
        at firrtl.passes.ZeroWidth$.firrtl$passes$ZeroWidth$$removeZero(ZeroWidth.scala:25)
        at firrtl.passes.ZeroWidth$$anonfun$3.apply(ZeroWidth.scala:19)
        at firrtl.passes.ZeroWidth$$anonfun$3.apply(ZeroWidth.scala:19)
        at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
        at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
```

However, if I run it without any flags, it successfully finishes. Go ahead and use that solution for now, and I'll take a look at the pass.

jackkoenig, 2017-06-12:
Fixed by https://github.com/freechipsproject/firrtl/pull/594

closed 2017-06-12 by jackkoenig