chipsalliance/firrtl #507: Rename memory instances to something more useful and unique

stevobailey, 2017-03-22:
Currently, inferred memory instances (both module names and instances names) have _ext appended to them. However, this name is not super useful, as ext is rather common (and even _ext). Can we rename this suffix to something more unique? I propose _funcmemext, short for functional memory external (I assume ext means external). Open to suggestions, though. 

@azidar @shunshou @ben-k @grebe

colinschmidt, 2017-03-22:
What do you want the suffix for?

ben-k, 2017-03-22:
I think the point is just that _ext is not very descriptive, given that the pass only touches SRAM blackboxes, and if you wanted to have some automation matching on only these modules, searching for _ext might return false positives.  This makes sense to me.

I find _funcmemext similarly difficult to grok, though - perhaps _memblackbox?

stevobailey, 2017-03-22:
The problem with _memblackbox is that some of these get mapped to FFs if they're small.

ccelio, 2017-03-22:
I'm not familiar with FIRRTL's abilities, but is this something that's more appropriate for annotations inserted into the Verilog code? I.e., instead of trying to match on names, instead provide some sort of Verilog comment/annotation/directive/decoration.

colinschmidt, 2017-03-22:
Yeah I was getting at was @ccelio is saying. If you are trying to find all the modules for something it might be do-able in firrtl where the names don't matter. Thats why I wanted the specific use case to see if it made sense as a firrtl thing.

stevobailey, 2017-03-22:
But if you're trying to find the memories in Design Compiler, the names
will be the only way to do it.

<https://github.com/ucb-bar/firrtl/issues/507#issuecomment-288523958>, or mute
the thread

colinschmidt, 2017-03-22:
Yeah I guess what I'm saying is it seems like if you want a list of all memories you should print them from the replseqmem pass or something rather than grepping through the source for same magic string later.

ben-k, 2017-03-22:
Fair enough.  It is still a code readability issue, though.

jackkoenig, 2017-12-15:
I don't think this should change. It's been the API for a long time so changing it could cause problems for people relying on the current extension. That being said, no one should be relying on these names--everyone should use the file emitted by ReplSeqMems. This could also perhaps be an option where the default behavior stays `_ext` and it can be optionally overwritten with an annotation/command-line options.

chick, 2018-12-17:
Not a resolvable issue. Use linking annotations

closed 2018-12-17 by chick