chipsalliance/firrtl #509: InferReadWrite fails with MatchError

edwardcwang, 2017-03-22:
FIRRTL commit 13fc27f35e85026c002e644b61c32268bd258d78

With a circuit like

```
module x:
  <...>
  module foo:
    <...>
  module bar:
    <...>
```

Running something like `firrtl.Driver -i input.fir -o input.v -X verilog --infer-rw foo` causes a MatchError, since the circuit main is "x" whereas the annotation that the command line interface creates has a circuit name of "foo", so it causes a MatchError.

Workaround is to just change the command-line interface call or hack the pass to run regardless of the circuit name.

Proposed solution: create another "InferReadWriteAnnotationModule(moduleName)" to handle this case, and change the command-line and/or barstools to use this.

https://github.com/ucb-bar/firrtl/blob/master/src/main/scala/firrtl/passes/memlib/InferReadWrite.scala#L163

azidar, 2017-12-15:
Will be fixed with annotation refactor

jackkoenig, 2018-06-22:
Fixed by https://github.com/freechipsproject/firrtl/pull/721

closed 2018-06-22 by jackkoenig