chipsalliance/firrtl #512: Remove Chirrtl tries to reference memory signal before declaring it

jackkoenig, 2017-03-28:
The following Firrtl
```
circuit Test :
  module Foo :
    input clock : Clock
    input reset : UInt<1>
    input in : UInt<32>
    output out : UInt<32>

    reg cycle : UInt<8>, clock with : (reset => (reset, UInt(0)))
    cycle <= tail(add(cycle, UInt(1)), 1)

    smem mem : UInt<32>[256]
    read mport mem_r = mem[cycle], clock
    write mport mem_w = mem[not(cycle)], clock

    mem_w <= in
    out <= mem_r

  module Test :
    input clock : Clock
    input reset : UInt<1>
    input in : UInt<32>[1]
    output out : UInt<32>[1]

    inst foo of Foo

    foo.clock <= clock
    foo.reset <= reset
    foo.in <= in[0]
    out[0] <= foo.out
```

Gets the following lines after Remove Chirrtl

```
    mem.mem_r.en <= UInt<1>("h1") @[:test.fir@10.4]
    mem mem : @[:test.fir@12.4]
```

jackkoenig, 2017-03-29:
Note that the following works:

```
circuit Test :
  module Foo :
    input clock : Clock
    input reset : UInt<1>
    input in : UInt<32>
    input en : UInt<1>
    output out : UInt<32>

    reg cycle : UInt<8>, clock with : (reset => (reset, UInt(0)))
    cycle <= tail(add(cycle, UInt(1)), 1)

    smem mem : UInt<32>[256]
    node raddr = cycle
    node waddr = not(cycle)
    write mport mem_w = mem[waddr], clock
    mem_w <= in
    read mport mem_r = mem[raddr], clock
    out <= mem_r

  module Test :
    input clock : Clock
    input reset : UInt<1>
    input in : UInt<32>[1]
    output out : UInt<32>[1]

    inst foo of Foo

    foo.clock <= clock
    foo.reset <= reset
    foo.in <= in[0]
    out[0] <= foo.out
```

albert-magyar, 2019-10-29:
Closed and absorbed into #727

closed 2019-10-29 by albert-magyar