chipsalliance/firrtl #519: Internal error using Driver.compile

colinschmidt, 2017-04-01:
I got a deprecation warning about this function so I'm going to try to fix that first but it seems like I'm supposed to always submit Internal Error issues.

```
[error] (run-main-0) firrtl.FIRRTLException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
firrtl.FIRRTLException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
        at firrtl.Utils$.error(Utils.scala:366)
        at firrtl.Utils$.throwInternalError(Utils.scala:132)
        at firrtl.Compiler$class.compile(Compiler.scala:369)
        at firrtl.VerilogCompiler.compile(LoweringCompilers.scala:135)
        at firrtl.Driver$.compile(Driver.scala:56)
        at barstools.tapeout.transforms.GenerateTopAndHarnessApp$class.firstPhase(Generate.scala:164)
        at barstools.tapeout.transforms.GenerateTopAndHarness$.firstPhase(Generate.scala:211)
        at barstools.tapeout.transforms.GenerateTopAndHarness$.delayedEndpoint$barstools$tapeout$transforms$GenerateTopAndHarness$1(Generate.scala:215)
        at barstools.tapeout.transforms.GenerateTopAndHarness$delayedInit$body.apply(Generate.scala:211)
        at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
        at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
        at scala.App$$anonfun$main$1.apply(App.scala:76)
        at scala.App$$anonfun$main$1.apply(App.scala:76)
        at scala.collection.immutable.List.foreach(List.scala:381)
        at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
        at scala.App$class.main(App.scala:76)
        at barstools.tapeout.transforms.GenerateTopAndHarness$.main(Generate.scala:211)
        at barstools.tapeout.transforms.GenerateTopAndHarness.main(Generate.scala)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:95)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:56)
        at java.lang.reflect.Method.invoke(Method.java:620)
        at sbt.Run.invokeMain(Run.scala:67)
        at sbt.Run.run0(Run.scala:61)
        at sbt.Run.sbt$Run$$execute$1(Run.scala:51)
        at sbt.Run$$anonfun$run$1.apply$mcV$sp(Run.scala:55)  
        at sbt.Run$$anonfun$run$1.apply(Run.scala:55)
        at sbt.Run$$anonfun$run$1.apply(Run.scala:55)
        at sbt.Logger$$anon$4.apply(Logger.scala:84)
        at sbt.TrapExit$App.run(TrapExit.scala:248)
        at java.lang.Thread.run(Thread.java:798)
java.lang.RuntimeException: Nonzero exit code: 1
```

jackkoenig, 2017-04-01:
What commit of Firrtl is this? `Compiler.scala:369` is a comment on master

colinschmidt, 2017-04-01:
ah I'm on update-rename2 sorry aac53a169938551e0be64f2d6785fa40d81233c1

jackkoenig, 2017-04-01:
I wonder how difficult it would be to add the git hash to throwInternalError

colinschmidt, 2017-04-01:
I converted it to Driver.execute and essentially got the same internal error. It was also during the match on finalState.emittedCircuitOption.
I'm thinking this has to do with annotations getting dropped or not getting dropped and I'm looking into it.

jackkoenig, 2017-04-05:
Fixed in master

jackkoenig, 2017-04-05:
via 3608401852baa18b4deaa22669529830b751901a

closed 2017-04-05 by jackkoenig