chipsalliance/firrtl #533: Verify that mux tree generation results in good QoR

aswaterman, 2017-04-12:
Right now we're generating code like ```addr == 3 ? elt3 : (addr == 2 ? elt2 : (addr == 1 ? elt1 : elt0))```.  Verify that Design Compiler does a good job with that, as compared to balanced trees (```addr[1] ? (addr[0] ? elt3 : elt2) : (addr[0] ? elt1 : elt0)```) and explicit decoding (```(addr == 3 ?
 elt3 : 0) | (addr == 2 ? elt2 : 0) | (addr == 1 ? elt1 : 0) | addr == 0 ? elt0 : 0)```).

ben-k, 2017-04-13:
Or perhaps even case statements.  There's some interesting reading on Solvnet about this (search for "Multiplexer Mapping and Optimization"), but I couldn't find anything conclusive.

aswaterman, 2017-04-13:
Interesting, I'll check that out.

shunshou, 2017-04-13:
Please let me know what the conclusion is on that. I've been wondering about similar things myself. In general, Chisel assumes that the tools are very good at performing optimizations. I think it'd be a useful case study to see how true that assumption is. If not, optimizations should be addressed at the Firrtl level.

ben-k, 2017-04-13:
If this sort of question (how to structure Verilog for better DC QoR) is of concern more generally, there's another document called "Coding Guidelines for Datapath Synthesis" that has lots of useful tidbits on the topic.

shunshou, 2017-04-13:
Hm I should take a look. So should everyone working on Chisel. It seems like tools are best dealing with Verilog templated in a certain way (say the BRAM stuff for FPGAs), and I'm not really sure that people are generally aware that that's the case. But then that begs the question: is it just the emitter that needs to be changed or do additional passes that improve QOR need to be inserted?

aswaterman, 2017-04-13:
I think most of this is outside the scope of Firrtl.  Folks writing Chisel at a fairly low level need to follow the guidelines in docs like Ben mentioned.  Since Chisel and Verilog are fairly close to 1:1 when writing combinational logic, the same guidelines generally apply.

Firrtl should do basic optimization when possible, but IMO generally shouldn't be used as a crutch.

The mux tree generation is of particular concern because it's Firrtl that's doing the code generation, not the Chisel writer.

shunshou, 2017-04-13:
Sure. I tend to hand optimize things myself based off of how I best think the tools will handle my code. Or, there isn't an easy way for the tools to know that % (mod) can be implemented one way vs. another since the code doesn't have designer intent embedded into it. I think there are some high level helper functions you can make to have some of this be easier for people. 

However, I've had people tell me that hand optimization is silly, because tools are good enough, and Chisel's supposed to abstract away these problems. Since you imply the opposite (which I'm totally fine with), I think that should be advertised i.e. if you want highly optimized code, you should stylize it in your code and not rely on Chisel. 

Mux tree generation is one, but I think all Chisel utilities should be double checked for QOR. 

I have yet to look at the doc, but: do tools do a better job optimizing things if you leave the code as case statements? Or do they do better if you actually make a balanced tree? Or if you cascade a bunch of stuff like in your example, will they realize that order shouldn't matter?

aswaterman, 2017-04-13:
I think it varies from issue to issue, and relying on synthesis tools is the right approach for some things (e.g., datapath synthesis with redundant forms, like computing a*b+c).

Agreed that all Chisel utils should be audited for QoR.

Not sure about the mux vs. case matter; I'm going to look into it.

aswaterman, 2017-04-13:
My experimentation suggests Firrtl's strategy is good for power-of-2-sized vecs.  For the same cycle time constraint, it's lower area than either alternative I mentioned.  Since it _looks_ like unoptimized code but performs better, this result strongly suggests that DC is successfully pattern-matching it.

I think we can do slightly better on the decoding side for non-power-of-2, though.  We are currently always returning the 0th element for out-of-bounds accesses, which is not optimal.  It's better to fill out the tree recursively and symmetrically.  For example, if you have an 11-element vec, the mapping is

0
1
2
3
4
5
6
7
8
9
10
10
8
9
10
10

The code for this is simple; see Rocket's Seq indexing code: https://github.com/ucb-bar/rocket-chip/commit/7a4105e064ed33612e8ed3b56115e8d249dddd9e#diff-12a48066201f1113607731638c500960R16

aswaterman, 2017-04-13:
(The alternative is to use case statements with a default of 'x for synthesis and $random for simulation, but that is really going down the rabbit hole...)

shunshou, 2017-04-13:
Is it? I like the case statement method if it is indeed more optimal...

aswaterman, 2017-04-14:
It's an experiment I still plan to run, but it probably isn't more optimal, since DC is picking up the current pattern and intelligently optimizing it.

Also, it is a very invasive change to Firrtl.

shunshou, 2017-04-14:
It is, but I'd hope people will consider supporting more than just power of 2 things (like I wonder how much the tool vendors care to handle those situations), which is useful for what I've been working on.

aswaterman, 2017-04-14:
It would be genuinely surprising if they do anything smarter than the reflective pattern I mentioned.  When there's more time, I'll verify that; but since DC is doing a good job already, this is a decidedly lower priority issue than I thought.

azidar, 2017-05-10:
I know @albert-magyar has been running some tests as well with regards to when/else emission from Chisel, perhaps he can comment.

jackkoenig, 2017-10-01:
I believe this was verified in merging https://github.com/freechipsproject/chisel3/pull/510

closed 2017-10-01 by jackkoenig