chipsalliance/firrtl #535: Write a pass to check clock domain crossing safety

ben-k, 2017-04-13:
This is a summary of the discussion at the Hurricane-2 debrief meeting.

This Firrtl pass would be for verification only (no netlist modification).  The goal is to check that every crossing between clock domains (that is, any combinational path that is bounded by registers with different clock inputs) uses particular modules annotated as "safe".  (In Rocket Chip, these modules would be LevelSyncCrossing and AsyncQueue.)  Any other unsafe crossings should cause an error.

shunshou, 2017-04-13:
You might have ad-hoc resynchronizers that people create with simple ShiftRegisters (say on reset). That should be considered too?

sdtwigg, 2017-04-13:
Honestly, those should be marked (annotated) anyways. Often you'll need to
treat them specially during PD and other verification anyway. Also, you may
need to swap them out with a 'better' version and the annotation would help
you find them.

aswaterman, 2017-04-13:
I agree with @sdtwigg - forbidding ad-hoc synchronization is the biggest benefit of this pass.  Even when ShiftRegisters are the best way to perform a CDC, they should be annotated as such.

shunshou, 2017-04-13:
I think that's reasonable.