chipsalliance/firrtl #539: Emit list of modules to retime

ben-k, 2017-04-13:
This is a summary of the discussion at the Hurricane-2 debrief meeting.

A Firrtl pass should emit a list of modules for which the designer intent is to retime them using the synthesis tools.  Such intent would be described in an annotation of the module in the Chisel source.

Whether this emits a simple list of names, or a properly-formatted (and potentially tool-specific) list of .tcl commands, is not decided.  But the designer intent should certainly be encoded in the source and passed through to the backend somehow.

colinschmidt, 2018-04-25:
See https://github.com/freechipsproject/rocket-chip/pull/1333

chick, 2018-12-17:
Consider in the future whether to move this to chisel without rocket

closed 2018-12-17 by chick