chipsalliance/firrtl #540: Support for clock muxes and clock dividers

ben-k, 2017-04-13:
This is a summary of the discussion at the Hurricane-2 debrief meeting.

Clock muxes and clock division registers should have first-class support in Firrtl.  Not sure if this needs an accompanying Chisel issue.

The related issue here is that the designer may wish to specify a library cell to use as a clock mux or divider (to ensure balanced rise/fall times on a clock signal, for example) rather than simply defining a mux construct in Verilog and giving the synthesis tools free reign.  It's unclear what the API for this would look like.

aswaterman, 2017-04-13:
What's wrong with using black boxes for this feature?  You can use syntactic sugar to make it look nice (e.g., ClockMux(Bool, Clock, Clock) can instantiate a black box).

shunshou, 2017-04-13:
I'm using the black box approach. I think it's 100% sufficient + some SDC constraint that only allows the tools to upsize/downsize the mux. There's a synthesis version that does sel ? opt1 : opt0 and there's a real implementation version that just uses a foundry mux. I don't think it needs to be a Firrtl pass; you just need to provide a utility in Chisel to support it -- (that's populated by some foundry-specific template -- so maybe a little bit of Firrtl magic involved).

ben-k, 2017-04-13:
@aswaterman - My only reluctance about blackboxing is that it might be nice for Chisel/Firrtl to still understand that the thing is a Mux.  I'm not certain about the resolution of ucb-bar/chisel3#551, but if Scala-based testing is still a goal, then blackboxes break that functionality.  Unless we can define functional behavior for blackboxes now?

shunshou, 2017-04-13:
Yea, I think you'd need to special case them for FirrtlInterpreter support. I haven't been using it for a while courtesy of limitations w/ multiclocking, etc., but it is useful for fast simulations. 

Is the point of FirrtlInterpreter to be fully compatible with all potential "advanced" features though? 

When thinking about physical design/larger scope chips, I think it's possible to request a lot of funky things that "dirty" the Firrtl codebase. People keep mentioning not wanting to bring the bad aspects of Verilog into Chisel/Firrtl, but some of these sketchy things -- like non-synthesizable Reals -- are extremely useful to mixed signal chips... Are we supposed to be special casing each instance of these sketchy but super useful features? Where's the boundary between Chisel/Firrtl + pure Verilog/SystemVerilog?

aswaterman, 2017-04-13:
For other reasons, black-box linkage will become a mandatory feature for Scala-based testing to be generally useful.  When using the Verilator backend, it should just be a matter of providing an additional list of Verilog sources.  It's a much harder problem for the Firrtl interpreter, where we'd need a functional Firrtl model of the black box.

sdtwigg, 2017-04-13:
Similar to the synchronizer discussion, I've been leaning towards the
blackbox solution for this issue (well, similar variants). The projected
plan for retaining FIRRTL analytical capabilities is annotations. The
general justification is similar to synchronizers where you generally want
to fine-tune the implementation for these things rather than letting the
synthesis engine do whatever.

This could be a good candidate for these sorts of blackboxes being
'grayboxes' similar to how LLVM has intrinsic functions. They would
basically blackboxes with spec-defined behavior. So, some passes could
EITHER treat them like any other blackbox OR try to interpret 'over them'
using the spec-defined behavior others (like when clock analysis). (I kind
of wanted mid-low firrtl memories to be like this.)

aswaterman, 2017-04-13:
The graybox thing is a little bit like how chisel2 blackboxes worked, where in the module body you'd write a behavioral description, used for the C++ backend.

chick, 2018-12-17:
This is good example of the utility of gray box concept