chipsalliance/firrtl #552: Approach for Providing Parameters to Custom FIRRTL Passes

seldridge, 2017-04-26:
Does there exist a method for communicating parameters to a custom FIRRTL pass or are these expected to be provided entirely by annotations?

My knee-jerk reaction is to add support for some command line argument string that is blindly given to all custom passes (or to the pass which preceded it). However, I think this can be equivalently done entirely with annotations.

Is there an advised approach here?

jackkoenig, 2017-04-28:
The advised approach is annotations. I know they're kind of verbose and clunky (if you've seen how in our Scala code we hide the fact that all Annotations are the same class). I'm planning to switch to using https://github.com/scala/pickling which would make all of this much cleaner, although there is some concern about the longterm maintenance of that project 👎

seldridge, 2017-04-28:
Thanks @jackkoenig. I'd come to a similar conclusion after digging around and actually understanding how the annotations work (the discussion here helped: https://github.com/ucb-bar/chisel3/issues/263). I had a fundamental misunderstanding in what the annotations were -- I thought these would be something inline in the emitted FIRRTL, but it's manifested as a separate file and input to FIRRTL. The annotations could be generated by Chisel, written explicitly, or emitted by some other tool and passed along to FIRRTL which does keep FIRRTL (very rightly) distinct from Chisel.

Having Chisel's Driver not emit annotations by default was a bit of a stumbling block (similarly with Rocket Chip's Generator), but I've at least gotten a hello world BlackBox working.

closed 2017-04-28 by seldridge