chipsalliance/firrtl #556: Array[Byte] should be purged from the FIRRTL AST

sdtwigg, 2017-05-02:
Array[Byte] as the representation of strings is problematic and should be replaced with String. Interestingly, my analysis was originally because I thought Array[Byte] could pose a performance issue; however, it turns out to be more-related to correctness instead.

A couple of reasons in roughly descending order of importance:
* Array[...] uses reference equality and bases the hashcode on the memory address of the Array. e.g. try ```Array(1,2,3) == Array(1,2,3)``` in the REPL. Note that Java Strings already cache their hashcode (except in the very rare case the hashcode ends up being exactly 0).
PS: Note that this relates to the finer point where, while an Array can be treated as (implicitly converted to) a Seq, it is not itself a Seq. e.g. try ```(Array(1,2,3): Seq[Int]) == (Array(1,2,3): Seq[Int])``` in the REPL.

* Array[Byte] is mutable. This is somewhat pedantic because you can potentially legislate with style guides; however, this mutability allows the writing of some potentially malformed passes, with particularly odd behavior in the case where they change one Array and that change propagates to multiple parts of the tree (since the same Array reference propagated).
Amusingly, due to the fact that Array does not have value equality or 'value hashcode' this still means it is safe to cache the hashcode of AST elements with an array in them. However, conversely, it is hard to reason about when AST nodes that serialize to the same thing actually end up being equalling as it matters whether the backing Arrays are actually the same reference.

* Array[Byte] is currently manifested by converting the (generally stored as UTF-16) String to UTF-8 representation and returning those bytes. In the face of non-ASCII characters (particularly if mixing classes of characters), there is no guarantee that a single byte actually represents a single Unicode codepoint (can be 1-4 bytes). The native String already handles this all internally. Regardless, the parser intake and emitter outtake are converting these Array[Byte] into String anyway (while also forcing a UTF-8 conversion), so you aren't really avoiding any weirdness there.

* Note that, if the space of a String versus Array[Byte] is a significant concern: while Strings are usually backed by a char[] in most JVMs, there are flags to tell the JVM to try to store the String as a byte[] array when the String contains conforming characters. That said, all these Array[Byte] are converted to String eventually during Verilog emission (or start as String during Verilog intake from the parser during visiting) so we would already notice if the use of String was exploding the memory footprint.

jackkoenig, 2017-05-02:
**Spoiler alert:**
```scala
scala> Array(1, 2, 3) == Array(1, 2, 3)
res0: Boolean = false

scala> (Array(1, 2, 3): Seq[Int]) == (Array(1, 2, 3): Seq[Int])
res1: Boolean = true
```

jackkoenig, 2017-09-30:
Fixed by #663

closed 2017-09-30 by jackkoenig