chipsalliance/firrtl #558: Add Support for Custom Command-Line Options

jackkoenig, 2017-05-03:
It would be really cool if transforms could register their own command-line options--at least boolean and single argument string options. I find that directory names and filenames are super common arguments to transformations and it would be great to be able to do this on the command-line.

I think the API for this is pretty simple: override a function that registers your commandline options. Any transformations passed to firrtl via the `-fct` option will have this function called (and possible any transforms instantiated via an annotation present in the annotation file). Firrtl would obviously check for no option collisions, although it would be cool if transformations could also *share* an option.

chick, 2017-05-03:
The options parsing occurs too early to allow the transforms to inject their own scopt style options

I think the only way to do this is to have something like -D"<sub-options>" where sub-options are in some proscribed form.  The options parsing could create a pool of these, probably as part of the FirrtlOptions,  then provide a simple API that would let transforms access their elements in the pool.
I think scopts only supports one occurrence of any given option so all the desired sub-options would have to be assembled by the invoker into a single -D (or whatever it ends up being called)

seldridge, 2019-04-26:
Fixed via #1005. 

An example is in the tests:
  - [src/test/scala/firrtlTests/options/RegistrationSpec.scala](https://github.com/freechipsproject/firrtl/blob/master/src/test/scala/firrtlTests/options/RegistrationSpec.scala)
  - [src/test/resources/META-INF/services/](https://github.com/freechipsproject/firrtl/tree/master/src/test/resources/META-INF/services)

closed 2019-04-26 by seldridge