chipsalliance/firrtl #562: Zero width Performance bug?

donggyukim, 2017-05-10:
I was compiling boom(https://github.com/ucb-bar/riscv-boom/tree/thesis) and got this:

```
======== Starting Transform ZeroWidth$ ========
[error] (run-main-0) java.lang.OutOfMemoryError: GC overhead limit exceeded
java.lang.OutOfMemoryError: GC overhead limit exceeded
	at scala.collection.TraversableLike$class.collect(TraversableLike.scala:282)
	at scala.collection.AbstractTraversable.collect(Traversable.scala:104)
	at firrtl.passes.ZeroWidth$.firrtl$passes$ZeroWidth$$removeZero(ZeroWidth.scala:18)
	at firrtl.passes.ZeroWidth$$anonfun$3.apply(ZeroWidth.scala:18)
	at firrtl.passes.ZeroWidth$$anonfun$3.apply(ZeroWidth.scala:18)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
	at scala.collection.AbstractTraversable.map(Traversable.scala:104)
	at firrtl.passes.ZeroWidth$.firrtl$passes$ZeroWidth$$removeZero(ZeroWidth.scala:18)
	at firrtl.passes.ZeroWidth$$anonfun$3.apply(ZeroWidth.scala:18)
	at firrtl.passes.ZeroWidth$$anonfun$3.apply(ZeroWidth.scala:18)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
	at scala.collection.AbstractTraversable.map(Traversable.scala:104)
	at firrtl.passes.ZeroWidth$.firrtl$passes$ZeroWidth$$removeZero(ZeroWidth.scala:18)
	at firrtl.passes.ZeroWidth$$anonfun$3.apply(ZeroWidth.scala:18)
	at firrtl.passes.ZeroWidth$$anonfun$3.apply(ZeroWidth.scala:18)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
	at scala.collection.AbstractTraversable.map(Traversable.scala:104)
	at firrtl.passes.ZeroWidth$.firrtl$passes$ZeroWidth$$removeZero(ZeroWidth.scala:18)
	at firrtl.passes.ZeroWidth$$anonfun$3.apply(ZeroWidth.scala:18)
	at firrtl.passes.ZeroWidth$$anonfun$3.apply(ZeroWidth.scala:18)
```

Is there any way to make it optional?

jackkoenig, 2017-05-10:
It should probably at least not bother running if there are no zero-width wires.

As a workaround, try increasing the heap size.

jackkoenig, 2017-09-30:
Pretty sure this was fixed by https://github.com/freechipsproject/firrtl/pull/594

closed 2017-09-30 by jackkoenig