chipsalliance/firrtl #591: Eliminate need for black_box_verilog_files.f

ucbjrl, 2017-05-19:
Track @shunshou's comment from #551 

> Personally, I prefer that whatever is in generated_src or test_run_dir/DUTblahblah is fully generated by Chisel (like all black box files should be moved there) and then just, for the simulator, do generated_src/*.v... Then you don't need the .f file...

aswaterman, 2017-05-19:
The .f file is still useful for other purposes.  I think we can just have it both ways?

aswaterman, 2017-05-19:
Also, if you forget to "make clean" first, then looking at generated-src/*.v is not necessarily right, as it may contain files from the old build that were not overwritten by the new build.  And if "make clean" is required, something is wrong :)

seldridge, 2018-12-18:
Closing for now. We can revisit this if people still need (don't need) it.

closed 2018-12-18 by seldridge