chipsalliance/firrtl #597: Registers which have dangling input generates incorrect verilog

jchang0, 2017-06-01:
Ticket migrated from:

https://github.com/ucb-bar/chisel3/issues/621

Correct behavior for register declaration in chisel is for register to keep the previous value even if the input is dangling.  Current verilog generation does not have that behavior, instead the register is taken out to be a wire, so in synthesis, the wire is always floating instead of able to keep the initial random value.

Relevent portion from the chisel thread:
Firrtl:

```
circuit RandTop :
  module RandTop :
    input clock : Clock
    input reset : UInt<1>
    output io : {out : UInt<32>}

    clock is invalid
    reset is invalid
    io is invalid
    reg rand : UInt<32>, clock @[Rand.scala 8:17]
    io.out <= rand @[Rand.scala 9:10]

```

Current Verilog:

```
module RandTop(
  input         clock,
  input         reset,
  output [31:0] io_out
);
  reg [31:0] rand$;
  reg [31:0] _GEN_0;
  assign io_out = rand$;
`ifdef RANDOMIZE
  integer initvar;
  initial begin
    `ifndef verilator
      #0.002 begin end
    `endif
  `ifdef RANDOMIZE_REG_INIT
  _GEN_0 = {1{$random}};
  rand$ = _GEN_0[31:0];
  `endif
  end
`endif
endmodule

```

Should have:

```
module RandTop(
  input         clock,
  input         reset,
  output [31:0] io_out
);
  reg [31:0] rand$;
  reg [31:0] _GEN_0;
  assign io_out = rand$;
`ifdef RANDOMIZE
  integer initvar;
  initial begin
    `ifndef verilator
      #0.002 begin end
    `endif
  `ifdef RANDOMIZE_REG_INIT
  _GEN_0 = {1{$random}};
  rand$ = _GEN_0[31:0];
  `endif
  end
`endif
  always @(posedge clock) begin
    rand$ <= rand$;
    // or perhaps
    rand$ <= _GEN_0;
  end
endmodule
```

aswaterman, 2017-06-01:
The always ```@(posedge clock)``` should actually contain no assignments at all, as the register is neither reset nor assigned.  The register should only be set in the initial block.

edwardcwang, 2017-06-01:
I thought that @jackkoenig said that the default next value of a `Reg` should be its current value?

aswaterman, 2017-06-01:
That's right, but ```always @(posedge clock) r <= r;``` is equivalent to no assignment at all.

jchang0, 2017-06-01:
I want to create an actual physical register (that happens to take its output and feedback into itself).

I thought 
`always @(posedge clock) r <= r`
 does that, whereas no assignment is just a wire without register.

aswaterman, 2017-06-01:
I don't see the difference between the two.  If the register feeds back into itself, then for all time it will preserve the value it was given in the initial block.  If the register has no assignments, then for all time it will preserve the value it was given in the initial block.  They're functionally identical, right?

edwardcwang, 2017-06-01:
It's perhaps possible that one would want a physical register (i.e. flip flop) to show up in the hardware for some reason.

aswaterman, 2017-06-01:
It'll get pruned in synthesis.

jchang0, 2017-06-01:
I just asked Han, he said that synthesis actually doesn't get rid of such a flop (synthesis cannot do that optimization safely).  However, he said that is a good idea for chisel to get rid of it since it makes no sense to leave the flop in there.  (However, getting rid of it creates issue for my formal environment since it's going to treat it as floating instead of constant value, but that discussion belongs to another thread).

The second thing I asked him is what synthesis would do in the floating wire case (in the case which we take out the flop).  He said that by default, synthesis tools will ground the wire then perform any optimization based on that tie off.  This behavior can be changed by the user.  Given this fact, perhaps it is best if we also, by default, initialize such floating wires to 0 instead of a random value so that the verilog it generate will match the synthesis behavior more closely.  In fact, Clifford recommended that we explicitly tie it to 0 rather than initialize it to random value (or initialize it to 0) to make sure that all the tools will interpret it the same way, and more importantly, ensure all tools matches synthesis behavior.

aswaterman, 2017-06-01:
@jchang0 if you explicitly self-assign the register in the Chisel code, do you get Verilog that works for your formal purposes?

jchang0, 2017-06-01:
Two separate issues with respect to formal:

Case 1. If we git rid of the flop and leave it floating, formal behavior will behave differently from simulation behavior (which is also different from synthesis behavior):
  a. Synthesis behavior (most cases, unless user override): Tie to 0
  b. Current simulation behavior: set to random value that stays the same throughout simulation (this may cause missed bugs)
  c. Formal behavior: set to random value that may change throughout the analysis (this will find bugs correctly, but introduces unnecessary complexity to the proof)

Best fix for this (IMO): tie it to 0 to match synthesis behavior, as we will actually simulate the case that's going on in silicon instead of with the random values.  This will also solve the formal behavior mismatch problem since it's explicitly tied down.

Case 2: I want to generate a random constant value for formal analysis purposes.
  I currently try to emulate this by trying to create a register that feeds back on itself.  In Chisel, I found some other code that happens to allow firrtl to keep this register as described here.
https://github.com/ucb-bar/chisel3/issues/576

  However, if we decide to go with the solution as described in case 1, it should also affect the "trick" I used.  In this case, the trick should not work anymore, and I need another way to force find a random constant value for formal analysis purposes.  (Maybe a blackbox with a feedback flop explicitly defined to get it out the chisel loop?)

aswaterman, 2017-06-01:
I just did a trial synthesis, and my claim is correct; design compiler *does* prune the register, and it is constant-propagated as 0; in other words, it is treated just like a floating wire.

    reg wtf;
    always @(posedge clock)
      wtf <= wtf;
    assign io_deq_valid = _GEN_61 | wtf;

Information: The register 'wtf_reg' is a constant and will be removed. (OPT-1206)

jchang0, 2017-06-01:
Can we explicitly tie it to 0 to get a more defined behavior across different tools (synthesis, simulation, formal) since it doesn't seem like synthesis will try to optimize it based on the fact that it's a don't care anyway.

jchang0, 2017-06-01:
Actually, maybe not and we can just create a firrtl pass instead.

jackkoenig, 2017-06-01:
If it has this large of an effect on verification it might be worth having a built-in option in Firrtl. It is easy enough to do as a custom pass; although, registers will still be randomized during initialization until they are reset. If we wish to change **that** behavior, changes to Firrtl are necessary.

aswaterman, 2017-06-01:
Registers should definitely still be randomized.

There is a legitimate question as to whether disconnected wires should be, though.  @jchang0 raises a good point that synthesis is more conservative than our simulation model by driving floating wires to 0.  We should seriously consider following suit.  Note, this will also improve simulation performance, possibly dramatically, as we can then constant-propagate code dependent on disconnected wires.

aswaterman, 2017-06-01:
Similarly, constant-propagating registers that are not reset and never assigned makes sense.

cliffordwolf, 2017-06-02:
> driving floating wires to 0. We should seriously consider following suit.

That would be great. It would also fix the 400+ warning messages I get from Yosys when processing rocket-chip. @aswaterman

aswaterman, 2017-06-02:
@jackkoenig sounds like we should investigate this.

jackkoenig, 2017-06-02:
Agreed, I'll look into it

jackkoenig, 2017-12-09:
Ever since #605, registers that are left unconnected or only connected to a single thing (which could be its synchronous reset value) are removed. If you want to get and hold a random value you could use a blackbox.

azidar, 2017-12-29:
Can I close this issue?

cliffordwolf, 2017-12-31:
Either current rocket (9305672, which is the current head of the rocket RVFI branch) is not using this version yet, or the issue is not completely resolved:

```
Warning: Wire RocketTile_rocket.\frontend.tlb._GEN_16 [0] is used but has no driver.
Warning: Wire RocketTile_rocket.\frontend.tlb.px_array [3] is used but has no driver.
Warning: Wire RocketTile_rocket.\frontend.tlb.px_array [2] is used but has no driver.
...
Warning: Wire RocketTile_rocket.\dcache.metaArb.io_in_4_bits_addr [2] is used but has no driver.
Warning: Wire RocketTile_rocket.\dcache.metaArb.io_in_4_bits_addr [1] is used but has no driver.
Warning: Wire RocketTile_rocket.\dcache.metaArb.io_in_4_bits_addr [0] is used but has no driver.
found and reported 74 problems.
```

I'm assuming it is the former (current rocket is not using this version yet). What would I need to do to make it use the current version of firrtl so I can verify that the issue is resolved?

jackkoenig, 2018-04-03:
I believe the issues linked by @cliffordwolf should be fixed by https://github.com/freechipsproject/firrtl/pull/776 in rocket-chip since https://github.com/freechipsproject/rocket-chip/pull/1311

jackkoenig, 2018-07-31:
I believe this is resolved by https://github.com/freechipsproject/firrtl/pull/776

closed 2018-07-31 by jackkoenig