chipsalliance/firrtl #598: Assignment of init random values for register

jchang0, 2017-06-01:


cliffordwolf, 2017-06-02:
> I'm not sure what the solution for that is, I suppose the work around is probably different for each simulator.

A big issue with `$random` is that the standard actually defines which RNG an implementation must use, and it's not a very good one. `$urandom` does not have that limitation and implementations that support it usually provide a better RNG for `$urandom`, but it's a SystemVerilog feature, so it's not available everywhere.

Simply running the results of `$random` though a simple hash function, and/or only using a few of the bits returned by the function can already improve the randomness drastically. For example taking the values returned by `$random` modulo a large prime and then only using the lower 16 bits:

    _GEN_0[15:0] = $random % 371064217;
    _GEN_0[31:16] = $random % 371064217;
    _GEN_0[47:32] = $random % 371064217;
    _GEN_0[63:48] = $random % 371064217;

> In fact, he pointed out that $random isn't very good at all:
http://svn.clifford.at/handicraft/2014/vlogrnd/bad_random.v

A few words explanation: This example demonstrates that whenever the bits `[5:0]` returned by `$random` are zero, the bits `[27:24]` will be zero as well. A good RNG should of course return bits that are all independent of each other.

cliffordwolf, 2017-06-02:
> For example taking the values returned by $random modulo a large prime and then only using the lower 16 bits.

Yeahh.. about that: Don't do it! I've been running that through dieharder and arguably the results are even worse than for just $random.

But the following seems to work pretty well. So far it passed the first 80 tests in dieharder and failed none. I'll post an update when dieharder has finished.

	function [31:0] myrand;
		input [31:0] v1;
		input [31:0] v2;
		begin
			myrand = v1;
			myrand = myrand ^ (myrand << 13);
			myrand = myrand ^ (myrand >> 17);
			myrand = myrand ^ (myrand << 5);
			myrand = myrand ^ v2;
			myrand = myrand ^ (myrand << 13);
			myrand = myrand ^ (myrand >> 17);
			myrand = myrand ^ (myrand << 5);
		end
	endfunction

	...
	_GEN_0[31:0] = myrand($random, $random);
	_GEN_0[63:32] = myrand($random, $random);

That's essentially abusing xorshift32 as a hashing function. It seems to create good random values. The advantage over using xorshift32 directly is that this solution requires no changes with respect to things like seeding because it still uses $random at its core.

cliffordwolf, 2017-06-02:
Here are the results from dieharder:

| Method | PASSED | WEAK | FAILED
|:---|---:|---:|---:|
| Just Verilog $random | 87 | 6 | 21 |
| Mod Prime, Lower 16 Bits | 12 | 8 | 94 |
| myrand() method | 113 | 1 | 0 |

My code and dieharder output (*.rpt) can be found here:  
http://svn.clifford.at/handicraft/2017/vlogrngdh/

jackkoenig, 2017-06-05:
I'm thinking that fixing this can go alongside making it easier for users to provide their own random implementations (many would like to use $urandom for example).

Perhaps we can emit by default a function similar to the above guarded with  something like ``` `ifndef CUSTOM_RANDOM ... `endif``` so that users can implement the function themselves if they so choose.

seldridge, 2018-12-18:
Close, can be overridden via @jackkoenig's approach.

closed 2018-12-18 by seldridge