chipsalliance/firrtl #600: Add Ability to Disable Combinational Loop Detection

jackkoenig, 2017-06-06:
This is problematic for people when there are false loops.

Perhaps it would be nice to add a bit-level combinational loop checker that verifies if a combinational loop is real or not.

sdtwigg, 2017-06-06:
Somewhat amusingly, this does match what verilator will complain about.

closed 2017-06-13 by jackkoenig

mysoreanoop, 2021-10-02:
Is there any update on this? Would the bit-level combination loop checker be implemented sometime? Cause it would be better to have the combinational loops checking on for genuine cases, right?

aswaterman, 2021-10-02:
The conservative algorithm is still in force. I could be wrong, but disabling it might cause problems downstream: later compiler passes might fail to terminate in the presence of cycles in the combinational logic graph.

Jack’s suggestion remains interesting, but to avoid Verilator false positives, we’d need to do more than bit-level analysis: we’d also need to transform the RTL, expanding UInts into Vecs, etc.  And we’d probably need to restrict the transformation to bitwise operators, else we’d be getting into the business of emitting gate-level netlists for arithmetic circuits.  It makes the status quo seem not so bad.

commit a151a6f6 (2017-06-12): Add option to disable combinational loop detection

Resolves #600
commit d7a1c897 (2017-06-12): Add option to disable combinational loop detection

Resolves #600
commit 541003c5 (2017-06-13): Add option to disable combinational loop detection

Resolves #600