chipsalliance/firrtl #611: DontTouch Problems

jackkoenig, 2017-06-21:
Opening this issue to track some problems I've noticed
1. #593 Dedup does not do any renaming
    - When a node marked dontTouch resides in a module that gets deduped, DCE errors saying it cant find the node marked dontTouch
    - Even when we fix dedup renaming, **should** Modules with nodes marked dontTouch be deduplicated? Perhaps they should only be deduped if all instances have the dontTouch annotation?
1. ConstProp and CSE do **not** respect dontTouch
    - This only effects nodes since those passes only remove nodes
    - It's also a little bit awkward because they don't delete nodes. They just remove uses of the nodes so that any marked dontTouch will remain (and reflect the same information) but won't actually be driving anything
1. We need visibility into optimization passes
    - "Where did my signal go?"

ben-k, 2017-06-22:
> Even when we fix dedup renaming, should Modules with nodes marked dontTouch be deduplicated? Perhaps they should only be deduped if all instances have the dontTouch annotation?

I think this behavior is reasonable.

> We need visibility into optimization passes

This would be a cool feature, although it seems like a lot of work.

mwachs5, 2017-06-22:
> We need visibility into optimization passes

Is there a way currently to at least get a dump of firrtl after each pass (if that question even makes sense)?

colinschmidt, 2017-06-22:
you can use trace log level to get a firrtl dump every pass. I'll wait for jack to post the command

jackkoenig, 2017-06-22:
Try `-ll trace` on the command-line. It's primarily useful on toy designs for debugging the compiler but occasionally if you're desperate to see what happened to a signal it can help

jackkoenig, 2017-06-22:
>>We need visibility into optimization passes

>This would be a cool feature, although it seems like a lot of work.

Checkout [branch see-dce](https://github.com/freechipsproject/firrtl/tree/see-dce) to see proposed visability into at least DCE.

I'm also thinking about a Chisel-exposed API to have a message printed every time the name of something marked changes, for example:
```scala
val foo = Wire(new Bundle {
  val a = UInt(32.W)
  val b = UInt(32.W)
})
// ...
trace(foo) // or something similar
```
During Firrtl run:
```
...
======== Starting Transform LowerTypes$ ========
MyModule.foo -> MyModule.foo_a, MyModule.foo_b
-------------------------------------------------
...
======== Starting Transform DeadCodeElimination ========
MyModule.foo_a DELETED
--------------------------------------------------------
```

I think a lot of the machinery already exists for this although initially it'll probably say that `MyModule.foo` was deleted in LowerTypes but also mention that `foo_a` and `foo_b` were created from it or something.

jackkoenig, 2017-06-22:
@ccelio Also mentioned the Chisel 2 behavior that DCEd ports would be visible in the resulting Verilog but commented out, we could probably implement similar behavior.

colinschmidt, 2017-06-22:
The trace chisel api seems pretty cool, although I'm not often asking questions it would provide answers to.

jackkoenig, 2017-06-22:
I see it as a designer debugging feature but is certainly not necessary to use. Making `dontTouch` work in CSE and ConstProp is more important in my opinion.

ben-k, 2017-06-22:
> @ccelio Also mentioned the Chisel 2 behavior that DCEd ports would be visible in the resulting Verilog but commented out, we could probably implement similar behavior.

Yes, please!  It would be great to recover this feature, at least as a (on-by-default?) option.

jackkoenig, 2017-06-22:
Actually tracing signals is almost more important for people new to Chisel since it would help them understand what the compiler is doing.