chipsalliance/firrtl #613: Unhelpful Error on Bad Bit Extract

jackkoenig, 2017-06-22:
```
circuit test :
  module test :
    input a : UInt<2>
    output b : UInt<1>
    b <= bits(a, 0) ; should be bits(a, 0, 0)
```
Gives
```
======== Starting Transform CInferTypes$ ========
[error] (run-main-13) java.lang.IndexOutOfBoundsException: 1
```

chick, 2019-06-10:
look at chisel and see what is causing this to get to firrtl

chick, 2019-06-24:
@azidar @jackkoenig 
Currently the following chisel code
```scala
class BitLopper extends RawModule {
  val a = IO(Input(UInt(2.W)))
  val b = IO(Output(UInt(1.W)))

  b := a(0)
}
```
generates 
```
circuit BitLopper : 
  module BitLopper : 
    input a : UInt<2>
    output b : UInt<1>
    
    node _T_4 = bits(a, 0, 0) @[BitSpec.scala 14:9]
    b <= _T_4 @[BitSpec.scala 14:5]
```
Which seems to be correct, I've tried a couple of other permutations. Any suggestions on what the chisel code should look like to trigger the offending case?

chick, 2019-06-24:
There are only two references to `BitsExtractOp` and both push two `ILit` args.
```
      pushOp(DefPrim(sourceInfo, Bool(), BitsExtractOp, this.ref, ILit(x), ILit(x)))
      pushOp(DefPrim(sourceInfo, UInt(Width(w)), BitsExtractOp, this.ref, ILit(x), ILit(y)))
```
The emitter code is generic, I don't see  a way that it can emit less than three total arguments. Maybe the problem firrtl snippet is a fossil of some sort.

albert-magyar, 2020-03-11:
Fixed by #1148 -- `CheckChirrtl` and `CheckHighForm` share checks, so more stuff gets caught.

albert-magyar, 2020-03-11:
This particular example actually relies on an additional fix from #1421. I verified that Jack's original example gives an `IncorrectNumConstsException` instead of an out-of-bounds access.
```
circuit test :
  module test :
    input a : UInt<2>
    output b : UInt<1>
    b <= bits(a, 0) ; should be bits(a, 0, 0)
```

closed 2020-03-11 by albert-magyar