chipsalliance/firrtl #618: How to handle absolute path annotations

jackkoenig, 2017-06-29:
Opening this issue to spark discussion.

How should we handle full path annotations? By that I mean `Top.a.b.c.foo` instead of a Module-relative annotation like `Top.foo`. I'm not sure if Chisel even emits these but they can be specified. I'm also pretty sure that most transforms do not handle them correctly if at all.

In most cases, I think we can convert absolute path annotations into Module-relative. Without deduplication, these are equivalent. With deduplication the question gets a little hairier. Perhaps Modules should not be deduped at all if their annotations are not identical, that would make the problem very simple and mean that we can always safely make the conversion.

Thoughts?

colinschmidt, 2017-06-29:
I think most often differing annotations implies the modules should not be de-duped. Only case I can think of where thats not going to be the case is if you were trying to include some physical design information as an annotation. For example, I have two cores that are identical and I want to place and route them once and then instantiate them but I want to mark each one with its top-left corner as a physical co-ordinate. I don't think this kind of annotation is reasonable but its an example I thought of.

edwardcwang, 2017-06-29:
What does it mean to de-duplicate in this case, anyhow? If it means that MyFooCore_1 and MyFooCore_2 get de-duped, what stops us from just instantiating two MyFooCore's (de-duped) and then just positioning them differently? Am a bit confused here.

colinschmidt, 2017-06-29:
Yeah. I'm saying there are alternative solutions to the core positioning problem that don't involve firrtl annotations.

seldridge, 2018-12-18:
Closed via Target.

closed 2018-12-18 by seldridge